platformio/platformio-vscode-ide #2208: Intellisense Index Rebuild runs forever

ManfredKao, 2020-10-31:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 5.0.2

### Description of problem
Intellisense Index Rebuild runs forever whenever I add my private repo to lib_deps in the platformio.ini file. I only noticed this issue after upgrading to version 5.0.1 and 5.0.2. I don't remember what version I was running before the upgrade. This issue does not appear if my private repo is replaced with a public one. I'm signed into my github account in visual studio code.

#### Steps to Reproduce

1. Create a new project
2. Add a private repo to lib_deps in the platformio.ini file. (I have my private repo on github)
3. Save the platformio.ini file

### Actual Results
It says PlatformIO: Intellisense Index Rebuild and the icon on the left is spinning forever preventing me from building or uploading code.

### Expected Results
Intellisense Index Rebuild to finish running in a timely manner.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nanoatmega328new]
platform = atmelavr
board = nanoatmega328new
framework = arduino
lib_deps =
    [git web url to my private repo]
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Additional info

ivankravets, 2020-10-31:
Does it work if you run it manually from [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) using the next command?

```
pio project init --ide vscode
```

ManfredKao, 2020-10-31:
Hey Ivan,

Thanks for the reply. I tried running it manually from CLI using the command you included and had the same result.
The steps I took:
-Clicked on the PlatformIO Core CLI tab
[image]
-Typed in pio project init --ide vscode
-This generated the PIO project files and directory with a blank platformio.ini file
-Copied and pasted the contents of the ini file above with the lib_deps pointing to my private repo
-Same result, Intellisense Index Rebuild is running forever

ManfredKao, 2020-11-07:
Is there anything else I can try?

ivankravets, 2020-11-09:
Could you try to build a project? Do you see any issues?

ivankravets, 2020-11-09:
```
lib_deps =
    [git web url to my private repo]
```

Sorry, this is an issue. It seems VSCode's GIT can't handle your private repo.

closed 2020-11-09 by ivankravets