platformio/platformio-vscode-ide #2209: Can't open PlatformIO project with PIO Home

cdmistman, 2020-10-31:
```
OS_MAKE_DIRS
Server error: {"type":"PermissionError","args":[13,"Permission denied"],"message":"[Errno 13] Permission denied: '/home/colton\\\\work\\\\project\\\\firmware\\\\device\\\\firmware'"}
```

Currently using PIO on VSCode, through WSL. Currently, PlatformIO is _only_ installed in WSL, meaning that the extension should only know about its Linux environment.

Steps to reproduce:
1. Install VSCode and WSL2 (I'm using Ubuntu 20.04) on Windows
2. Install WSL Extension for VSCode, open VSCode in WSL2 instance
3. Install PlatformIO extension
4. PIO Home -> attempt to open any PlatformIO project
5. When "No Items" appears in the finder, attempt to open the directory by walking the filepath using the "New" button, which will result in the error at the top for every subdirectory you attempt

kabir-plod, 2020-12-10:
It looks like `IS_WINDOWS` in [platformio-home/app/config.js](https://github.com/platformio/platformio-home/blob/fc0847e602fbf041e99afb384d4f0a16960e11e8/app/config.js#L17) is incorrect for WSL. I'm not sure what `navigator` refers to.

cdmistman, 2020-12-10:
I just checked [platform-vscode-ide/src/constants.js](https://github.com/platformio/platformio-vscode-ide/blob/develop/src/constants.js#L9) using the VSCode Extension Debugger, and it seems to me that this `IS_WINDOWS` is set correctly there (see screenshot).

@kabir-plod I think `navigator` refers to the **VSCode window** itself (see [MDN](https://developer.mozilla.org/en-US/docs/Web/API/NavigatorID/platform)), which is running on Windows, even though the extension is running in Linux. If that's the case, then I think it make sense to close this issue and re-open it in [platformio-home](https://github.com/platformio/platformio-home) since they aren't properly checking the execution environment; they're only getting the window's environment. Thanks for the debugging!

[image]

closed 2020-12-10 by cdmistman