platformio/platformio-vscode-ide #2214: Start PIO Home Server

SinglWolf, 2020-11-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Linux, 4.19.0-12-amd64, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    inject_contrib_pysite()
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 80, in inject_contrib_pysite
    contrib_pysite_dir = get_core_package_dir("contrib-pysite")
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/core.py", line 37, in get_core_package_dir
    pkg = pm.get_package(spec)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/meta.py", line 400, in __init__
    self.metadata = self.load_meta()
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/meta.py", line 430, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/developer/.platformio/penv/lib/python3.7/site-packages/platformio/package/meta.py", line 386, in load
    data = json.load(fp)
  File "/usr/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/developer/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/home/developer/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-20:
Please remove`/home/developer/.platformio` folder and restart VSCode

closed 2021-01-20 by ivankravets