platformio/platformio-vscode-ide #2215: bug: Cannot create a new project on fresh install on Windows 10 Home. "Could not initialize project: PIO Core Call Error: (...) UnicodeEncodeError: 'charmap' codec can't encode character '\\u0142' in position 14: character maps to <undefined>"

fopeczek, 2020-11-03:
I tried to test the viability of platformio on Windows for the upcoming workshop we organize. Documentation never mentioned it is Linux-only software, so I went ahead and tried it on a vanilla, freshly installed Windows 10. Unfortunately I became stuck at the creation of any new project. Can anyone help me?

To reproduce:

1. Install windows 10 Home N 64 bit. I installed Windows on VirtualBox.
2. Install all updates, until you arrive at Windows version 1909. 
3. Reboot the machine.
4. Install VS Code from the official MS site. My version is 1.50.1 (user setup)
5. In the VS Code install the latest version of the PlatformIO IDE (version 2.2.0)
6. Restart the VS Code after being instructed to do so.
7. Open VS Code, click on the PlatformIO icon which is last on the left bar.
8. Under Quick Access click "New Project"
9. Fill in the pop-up dialog that asks the project name, board and platform; e.g. with values "test", "Arduino Uno", "Arduino".

Then the error pops up. See the attached screenshot. Here is a full quote of the error message:

```
PIO Core Call Error: "The current working directory\n\nError: Traceback (most recent call last):\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n cli() # pylint: disable=no-value-for-parameter\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 153, in project_init\r\n click.secho(\" %s \" % project_dir, fg=\"cyan\", nl=False)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 98, in <lambda>\r\n click.secho = lambda *args, **kwargs: _safe_echo(1, *args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 91, in _safe_echo\r\n click_echo_origin[origin](*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\termui.py\", line 548, in secho\r\n return echo(message, file=file, nl=nl, err=err, color=color)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\utils.py\", line 272, in echo\r\n file.write(message)\r\n File \"C:\\.platformio\\python3\\lib\\encodings\\cp1252.py\", line 19, in encode\r\n return codecs.charmap_encode(input,self.errors,encoding_table)[0]\r\nUnicodeEncodeError: 'charmap' codec can't encode character '\\u0142' in position 14: character maps to <undefined>\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"
```
Here is the same message with `\r\n` replaced with new lines:
```
PIO Core Call Error: "The current working directory

Error: Traceback (most recent call last):
 File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main
 cli() # pylint: disable=no-value-for-parameter
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__
 return self.main(*args, **kwargs)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main
 rv = self.invoke(ctx)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke
 return super(PlatformioCLI, self).invoke(ctx)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke
 return _process_result(sub_ctx.command.invoke(sub_ctx))
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke
 return ctx.invoke(self.callback, **ctx.params)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke
 return callback(*args, **kwargs)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 21, in new_func
 return f(get_current_context(), *args, **kwargs)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\project.py\", line 153, in project_init
 click.secho(\" %s \" % project_dir, fg=\"cyan\", nl=False)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 98, in <lambda>
 click.secho = lambda *args, **kwargs: _safe_echo(1, *args, **kwargs)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 91, in _safe_echo
 click_echo_origin[origin](*args, **kwargs)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\termui.py\", line 548, in secho
 return echo(message, file=file, nl=nl, err=err, color=color)
 File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\utils.py\", line 272, in echo
 file.write(message)
 File \"C:\\.platformio\\python3\\lib\\encodings\\cp1252.py\", line 19, in encode
 return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\\u0142' in position 14: character maps to <undefined>

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
 `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
 https://docs.platformio.org/page/faq.html

* Report this problem to the developers
 https://github.com/platformio/platformio-core/issues

============================================================"
```
[image]

ivankravets, 2020-11-03:
Please open [PlatformIo CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type
```
pio upgrade --dev
```

Restart VSCode. Does it work now?

fopeczek, 2020-11-04:
It cannot possibly work, because `pio` was never put into the search path. But when I entered the only directory containing `pio.exe`:

```
C:\.platformio\penv\Scripts>pio.exe upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.0.3a2
Release notes: https://docs.platformio.org/en/latest/history.html
```

...and yes!!! Now it works!!! Thank you very, very much!!

closed 2020-11-04 by fopeczek