platformio/platformio-vscode-ide #2216: Start PIO Home Server

frankcannon, 2020-11-04:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.2.0
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\donci\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\donci\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\donci\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\donci\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\donci\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\donci\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\donci\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\donci\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\donci\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 33, in <module>
    import twisted
ModuleNotFoundError: No module named 'twisted'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\donci\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (c:\Users\donci\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\donci\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-04:
Please remove `C:\Users\donci\.platformio\packages\contrib-pysite` folder and restart VSCode.

closed 2021-01-04 by ivankravets

frankcannon, 2021-01-04:
ok