platformio/platformio-vscode-ide #2218: Start PIO Home Server

lukehodges, 2020-11-04:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Linux, 5.4.0-51-generic, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/commands/home/command.py", line 52, in cli
    inject_contrib_pysite()
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/core.py", line 80, in inject_contrib_pysite
    contrib_pysite_dir = get_core_package_dir("contrib-pysite")
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/core.py", line 37, in get_core_package_dir
    pkg = pm.get_package(spec)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/base.py", line 212, in get_installed
    spec = self.build_legacy_spec(pkg_dir)
  File "/home/luke/.platformio/penv/lib/python3.6/site-packages/platformio/package/manager/_legacy.py", line 26, in build_legacy_spec
    for name in os.listdir(pkg_dir):
PermissionError: [Errno 13] Permission denied: '/home/luke/.platformio/packages/framework-arduinoespressif32'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/luke/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/home/luke/.vscode/extensions/platformio.platformio-ide-2.1.3/node_modules/platformio-node-helpers/dist/index.js:1:878983)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-05:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2021-01-05 by ivankravets