platformio/platformio-vscode-ide #2226: PlatformIO Toolbar: "PlatformIO: Test” icon

minayaserrano, 2020-11-08:
Hi,

I have updated PlatformIO version (Core: 5.0.2, Home: 3.3.1) and now the “PlatformIO: Test” task icon does not appear in PlatformIO Toolbar.

I saw it was changed in this commit: https://github.com/platformio/platformio-vscode-ide/commit/d55cec6166e8c7e860fbdebf3605fbfb541baf19

That button was very useful for me. Is there any reason why it has been removed?

Would it be useful if I made a PR to restore this button?

noisymime, 2020-11-10:
+1 on this issue. 

Both the `Test` and the `Run Task` icons were removed, which means the only way to run the tests now is via the cmd line. Would be great to have these options returned to the GUI

ivankravets, 2020-11-12:
Have you seen [Project Tasks](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) explorer?

noisymime, 2020-11-12:
Thanks Ivan, I hadn't seen it through there. 

That does at least give an option through the GUI, but it's quite a number of extra steps. Is there any reason `Test` can't also be on the Status bar? There doesn't appear to be any shortage of room and whilst it will obviously be different for every user, I'd suspect it's more used than say the `Clean` option.

ivankravets, 2020-11-12:
> but it's quite a number of extra steps.

Could you explain to me these steps?

noisymime, 2020-11-12:
At a quick click count it was 5 clicks vs 1 to run the test:

1. Platform IO menu
2. Spin down the `env` I was interested in (Then scroll)
3. Spin down the `Advanced` section
4. Click test
5. Click the Explorer to get back to where I was previously

That's compared to the single click previously on the Status bar. It's not the end of the world of course, but it does feel like a step backwards.

ivankravets, 2020-11-12:
Aha, thanks! 

Have you seen "green" hint at https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks ?

1. Move tasks to Explorer view
2. Select the working environment

Please note that you can also set custom keybindings for the tasks.

P.S: Why we removed those icons? People claimed that PIO consumes a lot of space on the bottom bar because they use other extensions.

noisymime, 2020-11-17:
Thanks Ivan!

Not sure if it's something specific to my set up (VSCode on Mac) but items in the Task list are not draggable for me. File etc all work fine, but Tasks cannot be dragged around.

The key shortcut has worked though, thanks. 

> P.S: Why we removed those icons? People claimed that PIO consumes a lot of space on the bottom bar because they use other extensions.

Users, you can never make us all happy 😄

ivankravets, 2020-11-18:
> Users, you can never make us all happy

We try to balance and make happy the most of them 😊

------

Do you drag this header?

[image]

minayaserrano, 2020-11-19:
> P.S: Why we removed those icons? People claimed that PIO consumes a lot of space on the bottom bar because they use other extensions.

Hi, @ivankravets, 

I mostly wanted to know if there was any reason why the button had been removed, I understand the reason and I agree @noisymime "...you can never make us all happy", hahaha!

No worries, I think a custom keybinding will be the best option.

Thank you!

closed 2020-11-30 by minayaserrano

muexxl, 2021-03-19:
+1 also from my side . I miss the flask icon.