platformio/platformio-vscode-ide #2230: Platformio doesn't show up

traffic-light, 2020-11-11:
Hello everyone.
Since today I'm not able to use platformio.
It doesn't startup/ show on boot. Not the plaformio face in the bar and no controlbuttons at the bottom (ribon)

Specs:
Os: Arch linux with latest updates
VScode: 1.50.1
platform io version: 2.2 (installed manually because it's not available in the store anymore)

astrohome, 2020-11-12:
I can confirm the issue. I had it installed and now the error is:
"The below 1 extension(s) in workspace recommendations have issues: platformio.platformio-ide (not found in marketplace)"

I reinstalled VSCodium just to be sure that's not my local issue, and now I can't even find it in the marketplace.

traffic-light, 2020-11-12:
The marketplace is an pretty old issue.
It's strange that it's not there.

But that's not my biggest problem.
I've installed it manualy via the marketplace download page. (download vsix file and install that one with the extension manager)

The problem is that platformio isn't working at all

traffic-light, 2020-11-14:
After installing another version of vs code, the aur repository of arch linux, all my problems are solved.

So, it's not a problem of platformio. At least, that's what I think.

closed 2020-11-14 by traffic-light