platformio/platformio-vscode-ide #2231: Start PIO Home Server

ShaikhAP, 2020-11-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.0
  PIO IDE: v2.2.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\HP\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\HP\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\HP\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\HP\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\HP\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\HP\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\HP\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\HP\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 55, in cli
    from twisted.internet import reactor
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\reactor.py", line 39, in <module>
    default.install()
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 196, in install
    reactor = SelectReactor()
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\selectreactor.py", line 72, in __init__
    posixbase.PosixReactorBase.__init__(self)
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\base.py", line 571, in __init__
    self.installWaker()
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 284, in installWaker
    self.waker = self._wakerFactory(self)
  File "C:\Users\HP\.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 79, in __init__
    client.connect(server.getsockname())
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\HP\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (c:\Users\HP\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\HP\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/148

closed 2021-01-20 by ivankravets