platformio/platformio-vscode-ide #2233: Start PIO Home Server

lincvv, 2020-11-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.0
  PIO IDE: v2.2.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 54, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\user\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 33, in <module>
    import twisted
ModuleNotFoundError: No module named 'twisted'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-04:
Please remove `C:\Users\user\.platformio\packages\contrib-pysite` folder and restart VSCode.

lincvv, 2021-01-05:
> Please remove `C:\Users\user\.platformio\packages\contrib-pysite` folder and restart VSCode.

does not work

ivankravets, 2021-01-05:
Do you use 3rd party antivirus tools?

ivankravets, 2021-01-20:
Should be fixed in the latest PlatformIO Core 5.0.5. Please run in PlatformIO IDE terminal `pio upgrade --dev`. Restart VSCode.

closed 2021-01-20 by ivankravets