platformio/platformio-vscode-ide #2239: Start PIO Home Server

Vanthanh2509, 2020-11-19:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.1.0
  System: Linux, 5.4.0-51-generic, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:847:185)
	at b.set html [as html] (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:846:554)
	at u.newPanel (/home/thanh/.vscode/extensions/platformio.platformio-ide-2.1.0/dist/extension.js:1:26272)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async u.toggle (/home/thanh/.vscode/extensions/platformio.platformio-ide-2.1.0/dist/extension.js:1:25794)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-05:
Please upgrade the PlatformIO IDE extension in VSCode to the latest version. The issue is already fixed.

closed 2021-01-05 by ivankravets