platformio/platformio-vscode-ide #2241: Please always show "show running tasks" and count in lower left of window

paschlie, 2020-11-21:
Might it be possible to keep the "show running tasks" icon and count displayed in the lower left of the window between the "errors" and "warnings" icons and counts, and the "PlatformIO: Home" icon, even when the "running task" count is 0; because when a "build" or "upload" task in initiated, it causes the "running tasks" icon to show, and it shifts all the icons to its right, shifting the icon beneath the cursor, changing which icon will be selected upon click, depending on whether there are any tasks running or not.

For example, if I place the cursor over the "build" icon and click, with no tasks running, the "running tasks" icon appears and shifts the icons such that "home" ends up under the cursor, and if one moves the cursor over the newly repositioned "upload" icon, waiting for the result of the "build", but when it finishes, the cursor ends up over the "clean" icon when the build task if finished, which I end up mistakingly clicking, when intending to select "upload".

Overall, an action initialed by an icon click should not change the position of the icon clicked on.

ivankravets, 2021-02-02:
Sorry, there is no API in VSCode to override the original order of VSCode's icons in the status bar :(

closed 2021-02-02 by ivankravets

OSHChip-Owner, 2021-06-17:
Hi paschlie,
I have the exact same annoying problem, and have lost count of the
number of times I have completed a compile followed by an unintended
clean. Good news is I have a solution. My environment is on Windows,
but I suspect this should work regardless of the platform.

If VScode is just sitting idle and you hover over the icons you mention
and do a right click (context menu) a menu pops up with the
title "Hide Status Bar" and a bunch of things with a checkmark.

If you do the same thing, but while a task is running (a compile) then
there is an additional item in the list, "Running Tasks". Fortunately the
menu remains displayed even if the task completes.
(if you don't get the context menu displayed before the task completes,
you wont see the entry, so depending on how long your compiles take,
you may need to be fast with the clicks)
Anyway, once you see the menu with "Running Tasks", uncheck the entry.
Now when you do a compile, the "Running Tasks" does not get displayed,
and the icons don't shuffle around.

I have searched extensively in both the platformio install area:
   C:\Users\MyName\.platformio
and where VSCode is installed
    G:\visual_studio_code   (I do the unpacked .zip type installs)
and in my project directory
and in the Windows Registry
but I could not find where the switch for this is stored.

I did find the string "Running Tasks" in 3 places, but I don't know
what to do with that info:

In G:\visual_studio_code\resources\app\out\vs\workbench\workbench.desktop.main.js  on line 2516

In G:\visual_studio_code\resources\app\out\nls.metadata.json on line 14733
(this looks like it is the menu, but I didn't find where the check marks are stored)

In G:\visual_studio_code\resources\app\out\vs\workbench\workbench.desktop.main.nls.js   on line 48

Anyway, it fixed the underlying problem of shifting icons for me,
it wasn't like I ever had a use for the "Running Tasks" icon.

I hope this helps you too.

daverathbone, 2021-09-05:
Thanks can this be fixed its driving me mad  also!!!
You click upload with the tick the NOT NEEDED EVER task icon which shifts all  the icons about.  At least on Arduino IDE it static.
So then by mistake on your next upload you click Clean which is a long pain on ESP32.
This above all bugs  needs fixing