platformio/platformio-vscode-ide #2243: Unable to open `sym-linked` or `lib_extra_dirs` libraries using The Problem List

sticilface, 2020-11-21:
I have found this problem in relation to VSCode and other extensions but not PIO.  The VSCode people always point to the extension being the culprit. I had found an issue somewhere on PIO ( and have just searched for it for 30mins and cannot find it ) where you said that this is due to .ino file being used and not a .cpp file. I am using a .cpp file here and I still get it.

I am using
VSCode - Version: 1.51.1
PIO - PlatformIO 5.0.3
PIO IDE - v2.2.1

If I build a project with a lib symlinked in the ./lib folder I get the following error when I click on a compile “problem”.  The project and everything else is working just fine. 

[image]

If I make a copy of the whole library, in this case ESPManager, and place it in a folder which I then add to platform.ini as a lib_extra_dirs I get exactly the same error.

If I then copy the actual directory to the ./lib folder in project root then it works as expected.

Looking at the error message it is resolving the path incorrectly by adding
/Users/amelvin/git/MySONOFF
with
/Users/amelvin/git/PangolinMQTT/src/PANGO.cpp

when the first part is the project root and the second part is the full path of the symbolic or extra_dirs lib.

This is essentially a fresh install, VSCode + PIO IDE… and nothing else.

Below are two images from the same build of a project.  The same # pragma warning is present in the `problems` list and the build terminal output.  

[image]
If I select the problem -> I get the error message. 

[image]
If I cmd + select the filename in the terminal it opens correctly. 

Also discussed here: https://community.platformio.org/t/problems-found-in-libraries-included-as-symlinks-or-by-lib-extra-dirs-are-not-found/17307/3

sticilface, 2020-11-23:
For what it is worth if I open the correct file and then click on the problem, the window shuts and the can't find message pops up!

ivankravets, 2021-01-20:
Is it a VSCode bug?

sticilface, 2021-01-30:
I have searched their documentation and any references to this type of error they point straight back to the lib/module/plug in? 

Can you reproduce it? 

Andrew

closed 2021-04-07 by stale[bot]