platformio/platformio-vscode-ide #2248: Installation Manager

haepeto, 2020-11-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\TomHaepers\.platformio\.cache\tmp-2884ooZtLsIMJk78\virtualenv-16.7.7\virtualenv.py", line 2628, in <module>
    main()
  File "C:\Users\TomHaepers\.platformio\.cache\tmp-2884ooZtLsIMJk78\virtualenv-16.7.7\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\TomHaepers\.platformio\.cache\tmp-2884ooZtLsIMJk78\virtualenv-16.7.7\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\TomHaepers\.platformio\.cache\tmp-2884ooZtLsIMJk78\virtualenv-16.7.7\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\TomHaepers\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\TomHaepers\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\TomHaepers\\.platformio\\python37'
New python executable in C:\Users\TomHaepers\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\TomHaepers\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async t.value (c:\Users\TomHaepers\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:37257)
	at async Promise.all (index 0)
	at async c:\Users\TomHaepers\.vscode\extensions\platformio.platformio-ide-1.10.0\dist\extension.js:1:5902
  ```

windy54, 2020-12-01:
I think I am getting a similar error on MAC Catalina 10.5.7
I have changed the symbolic link to point to python 3.5
here is the output from a terminal
 ~ % python --version
Python 3.5.1

I notice there is a statement that it is looking for something in a python3.8 folder, hopefully I have highlighted it using markdown! could this be the problem?

error report is
 VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Darwin, 19.6.0, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/pioinstaller/__main__.py", line 122, in main
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/core.py", line 829, in __call__
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/core.py", line 782, in main
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/core.py", line 1236, in invoke
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/core.py", line 1066, in invoke
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/core.py", line 610, in invoke
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/click/decorators.py", line 21, in new_func
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/pioinstaller/__main__.py", line 52, in cli
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/pioinstaller/core.py", line 72, in install_platformio_core
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/pioinstaller/penv.py", line 52, in create_core_penv
  File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-coq_awct/tmpd_8fc30s/pioinstaller.zip/pioinstaller/python.py", line 170, in find_compatible_pythons
  File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File **"/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 1702, in _execute_child**
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: '/usr/local/bin/python2'
 -> Installer version: 0.3.5
Platform: macOS-10.15.7
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/stevengale/.platformio/python3/bin/python3
Creating a virtual environment at /Users/stevengale/.platformio/penv

	at /Users/stevengale/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/Users/stevengale/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2021-01-05:
@haepeto => Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213

@windy54 does `/Users/stevengale/.platformio/python3/bin/python --version` work for you?

windy54, 2021-01-05:
No it does not.
In fact if I type python in, that does not work either.
So it looks like I have lost my symbolic link, I know I changed that a while ago because it was pointing at python2 and not 3.
I will investigate further and get back to you.

windy54, 2021-01-05:
here is some output from my terminal window
`stevengale@stevens-MacBook-Air ~ % alias python='python3'
stevengale@stevens-MacBook-Air ~ % python
Python 3.9.0 (default, Nov 21 2020, 14:01:50) 
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
stevengale@stevens-MacBook-Air ~ % ./.platformio/python3/bin/python --version
zsh: no such file or directory: ./.platformio/python3/bin/python
stevengale@stevens-MacBook-Air ~ % python
Python 3.9.0 (default, Nov 21 2020, 14:01:50) 
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
stevengale@stevens-MacBook-Air ~ % 
stevengale@stevens-MacBook-Air ~ % 
stevengale@stevens-MacBook-Air ~ % 
stevengale@stevens-MacBook-Air ~ % 
stevengale@stevens-MacBook-Air ~ % 
stevengale@stevens-MacBook-Air ~ % python
Python 3.9.0 (default, Nov 21 2020, 14:01:50) 
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> 
stevengale@stevens-MacBook-Air ~ % ./.platformio/python3/bin/python --version
zsh: no such file or directory: ./.platformio/python3/bin/python
stevengale@stevens-MacBook-Air ~ % cd .platformio/python3/bin 
stevengale@stevens-MacBook-Air bin % ls
2to3			idle3			pip3.8			python3			python3.8-config
2to3-3.8		idle3.8			pydoc3			python3-config
easy_install-3.8	pip3			pydoc3.8		python3.8
stevengale@stevens-MacBook-Air bin % python --version
Python 3.9.0
stevengale@stevens-MacBook-Air bin % cd ~
stevengale@stevens-MacBook-Air ~ % ./.platformio/python3/bin/python3 --version
Python 3.8.4
`
So I am guessing that there should be a python executable in that folder?

ivankravets, 2021-01-05:
Could you run the installer script manually https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux ?

windy54, 2021-01-06:
stevengale@stevens-MacBook-Air ~ % wget https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -O get-platformio.py
zsh: command not found: wget

then curl gives

stevengale@stevens-MacBook-Air ~ % curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -o get-platformio.py
stevengale@stevens-MacBook-Air ~ % ls *.py
get-platformio.py
stevengale@stevens-MacBook-Air ~ % python3 get-platformio.py 
Installer version: 0.3.5
Platform: macOS-10.16
Python version: 3.9.0 (default, Nov 21 2020, 14:01:50) 
[Clang 12.0.0 (clang-1200.0.32.27)]
Python path: /usr/local/opt/python@3.9/bin/python3.9
Creating a virtual environment at /Users/stevengale/.platformio/penv
Traceback (most recent call last):
  File "/Users/stevengale/get-platformio.py", line 68, in <module>
    main()
  File "/Users/stevengale/get-platformio.py", line 60, in main
    bootstrap()
  File "/Users/stevengale/get-platformio.py", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/pioinstaller/__main__.py", line 122, in main
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/core.py", line 829, in __call__
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/core.py", line 782, in main
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/core.py", line 1236, in invoke
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/core.py", line 1066, in invoke
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/core.py", line 610, in invoke
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/click/decorators.py", line 21, in new_func
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/pioinstaller/__main__.py", line 52, in cli
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/pioinstaller/core.py", line 72, in install_platformio_core
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/pioinstaller/penv.py", line 52, in create_core_penv
  File "/Users/stevengale/.piocore-installer-lh272im1/tmp17xks5ij/pioinstaller.zip/pioinstaller/python.py", line 170, in find_compatible_pythons
  File "/usr/local/Cellar/python@3.9/3.9.0_2/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/local/Cellar/python@3.9/3.9.0_2/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/local/Cellar/python@3.9/3.9.0_2/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/Cellar/python@3.9/3.9.0_2/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1819, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: '/usr/local/bin/python2'
stevengale@stevens-MacBook-Air ~ % python
dyld: Library not loaded: /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.5/Resources/Python.app/Contents/MacOS/Python
  Reason: image not found
zsh: abort      python

I think I will see what happnes if I revert back to a bash? shell rather than zsh, although I think the default change to zsh was in the last OS update.

Steve

ivankravets, 2021-01-06:
Your macOS is broken. You have multiple Pythons in the system. See `/Library/Frameworks/Python.framework/Versions/3.5/Resources`.

It looks like you tried manually to install multiple of them. I recommend uninstalling Python 3.5 from the system.

closed 2021-01-06 by ivankravets

windy54, 2021-01-06:
I uninstalled all python versions and installed 3.9 and it works, visual code and atom,

thanks for help

ivankravets, 2021-01-06:
Happy coding with PlatformIO!