platformio/platformio-vscode-ide #2250: Start PIO Home Server

mgaman, 2020-12-02:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Webview is disposed
	at b.assertNotDisposed (c:\Users\david\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:847:185)
	at b.set html [as html] (c:\Users\david\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:846:554)
	at c.newPanel (c:\Users\david\.vscode\extensions\platformio.platformio-ide-1.10.0\dist\extension.js:1:26423)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async c.toggle (c:\Users\david\.vscode\extensions\platformio.platformio-ide-1.10.0\dist\extension.js:1:25945)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-05:
Please upgrade the PlatformIO IDE extension in VSCode to the latest version. The issue is already fixed.

closed 2021-01-05 by ivankravets