platformio/platformio-vscode-ide #2252: Building different project than focused one (multi-root workspace)

HamzaHajeir, 2020-12-03:
Hi

It’s a new behavior to me that hitting `ctrl+alt+b` (or clicking build button) builds only one project in the workspace, independent if I’m editing a file from another project in the workspace.

VSCode About : 
Version: 1.51.1 (user setup)
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-10T23:34:32.027Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18362

PlatformIO core : 5.0.4b1 
PlatformIO Home : 3.3.1

whittedb, 2020-12-07:
I have this issue too, but I've found a workaround.  If you open PlatformIO, under project tasks you'll see Default and then env:your_env.  The env:your_env is the project that gets built by default.  If you open the platformio.ini from a different project (and wait for the sync to complete), you'll see env:your_env change to env:your_new_env.  Now the default build will build the new environment.

HamzaHajeir, 2020-12-07:
Ah I see, this list is roughly new, And yes, Its sync time when appending the env:your_env is a bit annoying xD.

HamzaHajeir, 2020-12-12:
Now, Even if focused in one project And tasks appears are related to the focused project, It's building another project by default. 

I'm now working with pio commands.

biosmanager, 2021-01-18:
I have the same issue. The whole multi-project workflow is not very good at the moment.

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2320

closed 2021-01-20 by ivankravets