platformio/platformio-vscode-ide #2258: PlatformIO taints python environment in unrelated projects

septatrix, 2020-12-06:
When opening totally unrelated projects platformio still somehow sets a python virtual environment. Executing `pip3 --version` in a terminal results in `pip 20.2.4 from /home/septatrix/.platformio/penv/lib/python3.8/site-packages/pip (python 3.8)`.

This results in unexpected behaviour (unexpected package version, unable to `pip3 install --user ...` etc.)

septatrix, 2020-12-09:
I have tried reproducing this but was unable to do so. The bug may have potentially arisen from some weird edge case when updating the extension and restarting vscode or something similar. Should I be able to find a reproducible setup I will create a new issue and close this for now.

septatrix, 2020-12-16:
I have found a directory which, when opened, always loads the platformio environment even though it contains no platformio project. I will try track the error down and hope to provide a MWE soon.

septatrix, 2020-12-16:
Is there maybe a way to let PlatformIO say what the current project is or what projects it found?

ivankravets, 2021-01-05:
This is correct behavior if you clicked on the "terminal" icon on the bottom toolbar. This button setups a terminal with PlatformIO Core virtual environment. 

If you need to work with a clean macOS environment, please use Menu > Terminal > New Terminal

closed 2021-01-05 by ivankravets

septatrix, 2021-01-05:
But it does that even for some workspaces which do not contain a platformio project.

septatrix, 2021-01-05:
That also does not seem to resolve the issue

septatrix, 2021-01-07:
Okay after clearing the VSCode cache everything seems to work now. Thanks for pointing me towards that option. I still think it is weird that the behaviour was not consistent for different directories but now everything seems to work.

ivankravets, 2021-01-07:
Happy coding with PlatformIO!