platformio/platformio-vscode-ide #2260: ESP32CAM Fails at Linking firmware.elf

TackleBac, 2020-12-08:
I am using PlatformIO v2.2.1     I am trying to setup this project although i am gettng some errors,  other errors i was able to debug and find a solution, but this elf issue I cannot seem to resolve.

The tutorial is here [https://d268s23yov0ww.cloudfront.net/aws-csdk-mqtt-rekognition-src.zip](url)  - a aws guide by amazon.

my plantformio.ini looks like this 

`[env:esp32dev]
platform = espressif32    //I have also tried using the url here as per other solutions, still same error
framework = espidf        // I have also swapped board and framework order, still same issue.
board = esp32cam
monitor_speed = 115200
upload_port = COM32
board_build.embed_txtfiles = 
	src/certs/private.pem.key
	src/certs/certificate.pem.crt
	src/certs/aws-root-ca.pem
board_build.partitions = partitions_singleapp.csv`

Below is the error i receive
`Linking .pio\build\esp32dev\firmware.elf
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\aws-root-ca.pem.S.o: in function `aws_root_ca_pem':
(.rodata.embedded+0x0): multiple definition of `aws_root_ca_pem'; .pio\build\esp32dev\aws-root-ca.pem.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\aws-root-ca.pem.S.o: in function `aws_root_ca_pem':
(.rodata.embedded+0x0): multiple definition of `_binary_aws_root_ca_pem_start'; .pio\build\esp32dev\aws-root-ca.pem.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\aws-root-ca.pem.S.o: in function `_binary_aws_root_ca_pem_end':
(.rodata.embedded+0x4a5): multiple definition of `_binary_aws_root_ca_pem_end'; .pio\build\esp32dev\aws-root-ca.pem.o:(.rodata.embedded+0x4a5): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\aws-root-ca.pem.S.o: in function `_binary_aws_root_ca_pem_end':
(.rodata.embedded+0x4a5): multiple definition of `aws_root_ca_pem_length'; .pio\build\esp32dev\aws-root-ca.pem.o:(.rodata.embedded+0x4a5): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\certificate.pem.crt.S.o: in function `certificate_pem_crt':
(.rodata.embedded+0x0): multiple definition of `certificate_pem_crt'; .pio\build\esp32dev\certificate.pem.crt.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\certificate.pem.crt.S.o: in function `certificate_pem_crt':
(.rodata.embedded+0x0): multiple definition of `_binary_certificate_pem_crt_start'; .pio\build\esp32dev\certificate.pem.crt.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\certificate.pem.crt.S.o: in function `_binary_certificate_pem_crt_end':
(.rodata.embedded+0x4c9): multiple definition of `_binary_certificate_pem_crt_end'; .pio\build\esp32dev\certificate.pem.crt.o:(.rodata.embedded+0x4c9): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\certificate.pem.crt.S.o: in function `_binary_certificate_pem_crt_end':
(.rodata.embedded+0x4c9): multiple definition of `certificate_pem_crt_length'; .pio\build\esp32dev\certificate.pem.crt.o:(.rodata.embedded+0x4c9): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\private.pem.key.S.o: in function `private_pem_key':
(.rodata.embedded+0x0): multiple definition of `private_pem_key'; .pio\build\esp32dev\private.pem.key.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\private.pem.key.S.o: in function `private_pem_key':
(.rodata.embedded+0x0): multiple definition of `_binary_private_pem_key_start'; .pio\build\esp32dev\private.pem.key.o:(.rodata.embedded+0x0): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\private.pem.key.S.o: in function `_binary_private_pem_key_end':
(.rodata.embedded+0x690): multiple definition of `_binary_private_pem_key_end'; .pio\build\esp32dev\private.pem.key.o:(.rodata.embedded+0x690): first defined here
c:/users/imgreenman/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\esp-idf\.pio\build\esp32dev\private.pem.key.S.o: in function `_binary_private_pem_key_end':
(.rodata.embedded+0x690): multiple definition of `private_pem_key_length'; .pio\build\esp32dev\private.pem.key.o:(.rodata.embedded+0x690): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`

I created new certs, does not make a difference, I have commented them out in the platfromio.ini to receive a different warning.  

There was a an error
`home/xxx/workspace/m5cam/components/esp32-camera/driver/twi.c:61:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;`

Which was resolved by [https://user-images.githubusercontent.com/30533684/82684064-04b3cc00-9c20-11ea-82bb-a4b4960fbe71.png](url)

Any help would be appriciated.

dmitry-cherkas, 2020-12-08:
+1, however the issue is quite unlikely to be related to the PlatformIO itself

kennycoder, 2020-12-09:
I'm building with the following versions:
` - espressif32 1.12.3
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)`

However after upgrading to the latest versions of everything it started to fail linking. I will look into what is happening and will report back here.

kennycoder, 2020-12-09:
https://community.platformio.org/t/m5stack-esp32-linking-error-collect2-exe-error-ld-returned-1-exit-status/17200/4

TackleBac, 2020-12-09:
> https://community.platformio.org/t/m5stack-esp32-linking-error-collect2-exe-error-ld-returned-1-exit-status/17200/4

Thank you very much it seems an espressif32 error??

dmitry-cherkas, 2020-12-09:
@kennycoder , thanks for the hint. Indeed when I downgrade the platform version in `platformio.ini` per below, the build passes successfully:
```
platform = espressif32@1.12.3
```

dmitry-cherkas, 2020-12-09:
Similar issue occurred in the past :https://github.com/platformio/platform-espressif32/issues/259

kennycoder, 2020-12-09:
Ok I found the issue with the newer version.
In the src folder, open CMakeLists.txt and remove the following:

if(CONFIG_EXAMPLE_EMBEDDED_CERTS)
target_add_binary_data(${COMPONENT_TARGET} "certs/aws-root-ca.pem" TEXT)
target_add_binary_data(${COMPONENT_TARGET} "certs/certificate.pem.crt" TEXT)
target_add_binary_data(${COMPONENT_TARGET} "certs/private.pem.key" TEXT)
endif()

This is not necessary anymore

TackleBac, 2020-12-09:
> Ok I found the issue with the newer version.
> In the src folder, open CMakeLists.txt and remove the following:
> 
> if(CONFIG_EXAMPLE_EMBEDDED_CERTS)
> target_add_binary_data(${COMPONENT_TARGET} "certs/aws-root-ca.pem" TEXT)
> target_add_binary_data(${COMPONENT_TARGET} "certs/certificate.pem.crt" TEXT)
> target_add_binary_data(${COMPONENT_TARGET} "certs/private.pem.key" TEXT)
> endif()
> 
> This is not necessary anymore
> 
> I will update the main source code provided with AWS's example.
> Thanks for reporting!

Brilliant Work, I havent really used CMAKE previously although some forum posts I found did mention CMAKE. I find alot of AWS tutorials outdated from either the way AWS layout of services, or code not building. Its great to be able to test this when i get home :)  Thank you

TackleBac, 2020-12-09:
> Ok I found the issue with the newer version.
> In the src folder, open CMakeLists.txt and remove the following:
>
> if(CONFIG_EXAMPLE_EMBEDDED_CERTS)
> target_add_binary_data(${COMPONENT_TARGET} [...]

Any idea why this appears?? 
\.platformio\packages\framework-espidf\components\mbedtls\port\esp32\esp_bignum.c:254:5: warning: no previous prototype for 'esp_mpi_mul_mpi_mod' [-Wmissing-prototypes]
 int esp_mpi_mul_mpi_mod(mbedtls_mpi *Z, const mbedtls_mpi *X, const mbedtls_mpi *Y, const mbedtls_mpi *M)

dmitry-cherkas, 2020-12-09:
@TackleBac , try starting all over. I've managed to make it work without any code changes (on `platform = espressif32@1.12.3`).
I had to update the IAM Policy to use wildcard for `iot:Subscribe` action, so replacing:
```
{
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:REGION:ACCOUNT_ID:topicfilter/esp32/sub/data"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:REGION:ACCOUNT_ID:topicfilter/esp32/sub/url"
    }
```
with
```
{
      "Effect": "Allow",
      "Action": [
        "iot:Subscribe"
      ],
      "Resource": [
        "arn:aws:iot:eu-central-1:466403372688:topicfilter/esp32/sub/*"
      ]
    },
```

Otherwise it would not subscribe to `esp32/sub/+` with `-12` error code. @kennycoder , you might want to incorporate this into the article as well. Thanks again for your help and putting this together, very useful.

TackleBac, 2020-12-10:
When i download the code form the tutorial and just build it, after running menu to setup ssid and password,  It throws up some warning about the CMAKE file is missing, also  "tcpip_adapter_init () is deprecated please use esp_netif_init ()"  do you change this ???  I have tried both ways and cannot seem to get the picture to send.   Running the test on iot core does return what the image objects are, just not successful from esp32cam as of yet.

dmitry-cherkas, 2020-12-10:
Deprecation warnings can be ignored, but if I recall correctly, I was only facing them on latest platform and not when I built with `platform = espressif32@1.12.3`. As for CMAKE errors, I was occasionally receiving those, cleaning the project and re-building it once again solved it for me.

TackleBac, 2020-12-10:
Ok I will try it later and report back what happens.