platformio/platformio-vscode-ide #2263: Community feedback

ivankravets, 2020-12-09:
- [ ] When opening a project, change the mode of VSC to open the control-shift-p menu with options to build, run, etc.
- [ ] When opening a project, instead of simply closing the PlatformIO window and reverting to the last open file (which would be the VSCode home for new window), it should open the PlatformIO.ini file. Otherwise it wasn’t obviously apparent at first glance that the project was successfully opened.

septatrix, 2021-01-07:
Regarding the first I would strongly advise against changing the behaviour of well know menus/shortcuts. Instead the usage of <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>B</kbd> should be encouraged (tasks menu).

closed 2021-02-02 by ivankravets

Taomyn, 2021-03-08:
Can we have a way to override this? Ever since now when Github Desktop opens my Marlin project in VScode the file is opened - I have zero use for this.

GsicariG, 2021-03-08:
> Can we have a way to override this? Ever since now when Github Desktop opens my Marlin project in VScode the file is opened - I have zero use for this.

I second this.

commit 39a53644 (2021-02-02): Open "platformio.ini" configuration file from newly added project (if there are no other active editors) // Resolve #2263