platformio/platformio-vscode-ide #2265: Start PIO Home Server

Toperko, 2020-12-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Darwin, 20.1.0, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1256, in invoke
    Command.invoke(self, ctx)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 63, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/maintenance.py", line 47, in on_platformio_start
    after_upgrade(ctx)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/maintenance.py", line 169, in after_upgrade
    update_core_packages(silent=True)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/core.py", line 49, in update_core_packages
    pkg = pm.get_package(spec)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py", line 212, in get_installed
    spec = self.build_legacy_spec(pkg_dir)
  File "/Users/tp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/_legacy.py", line 26, in build_legacy_spec
    for name in os.listdir(pkg_dir):
PermissionError: [Errno 13] Permission denied: '/Users/tp/.platformio/packages/contrib-piohome'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /Users/tp/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/Users/tp/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-04:
Please remove `/Users/tp/.platformio/packages/contrib-piohome` folder and restart VSCode.

closed 2021-01-04 by ivankravets