platformio/platformio-vscode-ide #2272: Start PIO Home Server

jeronimo2020, 2020-12-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 6.3.9600, x64

  %23 Exception
  ```
  Error: El sistema no puede encontrar la ruta especificada.
Error: spawn C:\Users\JERONIMO\.platformio\penv\Scripts\python.exe ENOENT
	at c:\Users\JERONIMO\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at c (c:\Users\JERONIMO\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.<anonymous> (c:\Users\JERONIMO\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879011)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\JERONIMO\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:30:37)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
  ```

ivankravets, 2020-12-17:
Do you have unofficial antivirus tools in the system?

hb020, 2020-12-24:
Potential duplicate of #148

closed 2021-01-04 by ivankravets