platformio/platformio-vscode-ide #2276: Show active project in statusbar and PlatformIO project tasks list

maxgerhardt, 2020-12-19:
(Feature Request, copied from https://community.platformio.org/t/feature-request-show-active-project-in-statusbar-and-platformio-project-tasks-list/17897)

When using VSCode workspaces with multiple projects, and having file sopen from 2 or more of the included oflders, it is often very unclear what project is the “Actve project” that will be build when using the statusbar icons or using the differnt options in the Project Tasks tree.

It would be really helpfull to show the active project, and maybe even select the active project like you now can select the active environment to build

Bascy, 2020-12-19:
Thank you @maxgerhardt  for posting it here

closed 2021-02-09 by ivankravets

ivankravets, 2021-02-09:
Please re-test with https://community.platformio.org/t/platformio-ide-for-vscode-2-3-beta-help-us-with-testing/19219

commit 0dc9bdeb (2021-02-09): Show active project in statusbar // Resolve #2276