platformio/platformio-vscode-ide #2278: Start PIO Home Server

KIM-PUSS, 2020-12-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Error: Could not install package 'platformio/contrib-pysite @ ~2.38.0' for 'windows_amd64' system

	at D:\Users\Admin\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (D:\Users\Admin\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (D:\Users\Admin\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Potential duplicate of #148

ivankravets, 2021-01-04:
Thanks, @hb020 . Yes, this is duplicate of https://github.com/platformio/platformio-vscode-ide/issues/148

It seems 3rd party antivirus software blocks PlatformIO.

closed 2021-01-04 by ivankravets