platformio/platformio-vscode-ide #2279: Start PIO Home Server

AsadR110, 2020-12-22:
%23 Description of problem
I'm trying to use platform i.o with my windows 10 pc but I am getting the following errors. 

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Error: The directory `C:\Users\Asad\.platformio` or its parent directory is not owned by the current user and PlatformIO can not store configuration data.
Please check the permissions and owner of that directory.
Otherwise, please remove manually `C:\Users\Asad\.platformio` directory and PlatformIO will create new from the current user.

	at c:\Users\Asad\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:5625
	at ChildProcess.c (c:\Users\Asad\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Asad\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

hb020, 2020-12-24:
Duplicate of #148

ivankravets, 2021-01-04:
The directory `C:\Users\Asad\.platformio` or its parent directory is not owned by the current user and PlatformIO can not store configuration data.
Please check the permissions and owner of that directory.
Otherwise, please remove manually `C:\Users\Asad\.platformio` directory and PlatformIO will create new from the current user.

closed 2021-01-04 by ivankravets