platformio/platformio-vscode-ide #2287: [FEATURE REQUEST] Options merge through extends

turgu1, 2020-12-30:
Version: PlatformIO IDE V2.2.1
Editor: VSCode
OS: Linux

Hello, it would be beneficial that the following platformio.ini would merge options, such that in my (over simplified) example below, for [env:device1], there would be two entries in the build_flags option, instead of one (with the current version of platformio, builds_flags in [env:device1] replaces the one from devices. I was expecting that they would have been merged):

```
[devices]
build_flags =
  -D MAIN_DEFINE=2

[env:device1]
extends = devices
build_flags =
  -D OTHER_DEFINE=1
```

Here is a real example that would benefit from this  feature (don't tell me to use [env] for this, the platformio.ini contains a lot more that can't rely on [env]):

```
[platformio]
default_envs = inkplate_6

[inkplate_common]
platform               = espressif32
board                  = esp-wrover-kit
framework              = espidf
monitor_speed          = 115200
upload_speed           = 256000
monitor_filters        = colorize
board_build.partitions = partitions.csv
build_flags =
  -O3
  -D DEBUGGING=0
  -D CONFIG_SPIRAM_CACHE_WORKAROUND
  -D EPUB_INKPLATE_BUILD=1
  -D EPUB_LINUX_BUILD=0
  -D INCLUDE_vTaskSuspend=1
  -D SHOW_TIMING=0
  !/usr/bin/pkg-config --cflags --libs lib_freetype/lib/pkgconfig/freetype2.pc
  -std=gnu++17
  -I include_global
lib_extra_dirs =
  lib_esp32
board_build.embed_files =
  embed/favicon.ico
  embed/upload_script.html
lib_deps =
  https://github.com/turgu1/ESP-IDF-InkPlate.git

[env:inkplate_6]
extends     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_6PLUS

[env:inkplate_6plus]
extends     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_6PLUS

[env:inkplate_10]
extends     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_10

```

Did I miss something from what is currently possible with the platform.ini structure? Is there another way to do it without compromising the clarity of the ini file? 
Thanks a lot for your help!!

Guy

ivankravets, 2021-01-04:
Hello,

You are looking for `extends` option, see https://docs.platformio.org/en/latest/projectconf/section_env_advanced.html

turgu1, 2021-01-04:
I'm sorry Ivan, but extends doesn't do the job. Please look closer to my simple example: If I do what you said:

```
[devices]
build_flags =
  -D MAIN_DEFINE=2

[env:device1]
extends = devices
build_flags =
  -D OTHER_DEFINE=1
```

In [env:device1], I loose the MAIN_DEFINE definition. Hope this is clearer.

Thanks!
Guy

ivankravets, 2021-01-04:
See [Dynamic variables](https://docs.platformio.org/en/latest/projectconf/dynamic_variables.html):

```
[devices]
build_flags =
  -D MAIN_DEFINE=2

[env:device1]
extends = devices
build_flags =
  -D OTHER_DEFINE=1
  ${devices.build_flags}

```

turgu1, 2021-01-04:
That is exactly what I want to avoid. When you have several hundred
lines of platformio.ini file, it become quiet combersome.