platformio/platformio-vscode-ide #2294: Button Bar disappeared

MonsterThews, 2021-01-03:
The Build, Clean, Upload, etc. buttons at the bottom of the PIO screen went missing for a few days.
There were no changes when it started.

I removed all the folders from the workspace, then added a new one.
Support guys said to be sure an .ini file was in the working directory, and there was.

I restarted VSCode several times, trying to vary something to get it to change. No menu items helped. I don't know command prompt stuff.

I opened it this morning, and the buttons are there. They were missing from Friday afternoon to Sunday morning.

ivankravets, 2021-01-04:
Do you see the "Home" button?

ivankravets, 2021-01-20:
Please reopen if you still need a help

closed 2021-01-20 by ivankravets