platformio/platformio-vscode-ide #2295: Disabling builtin python has no effect

blazoncek, 2021-01-05:
Running PlatformIO on VSCode for Windows does not honour disabling "Use builtin python".

Even with Python 3.9 installed on the computer (C:\Program Files\Python\Python39) PlatformIO continues to use builtin Python located in %homepath%\.platformio\penv\Scripts.

If the Microsoft Windows AppLocker is enabled by Group Policy to prevent running applications outside of permitted folders (e.g. C:\Program Files) this prevents using Python from within PIO.
[image]

[image]

ivankravets, 2021-01-05:
It means that PlatformIO Core installer used YOUR local Python to create virtual Python environment in `%HOME_DIR%/.platformio/penv` folder.

You can check it with running 
```
%HOME_DIR%/.platformio/penv/Scripts/python --version
```

closed 2021-01-05 by ivankravets

blazoncek, 2021-01-05:
The problem is that, due to company policies, applications outside of C:\program Files are not allowed to run.
The above command you provided just fails. Python is not alloved to run.

ivankravets, 2021-01-05:
You can override PLATFORMIO_CORE_DIR, see https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR