platformio/platformio-vscode-ide #2297: program stucked when creating a new project or adding libraries to a project

James-C-Hunt, 2021-01-05:
I am an engineer in China .When I create a new project  or add libraries to my project  using the platformIO, the program stuck and told me to wait for moment.shown as fellow.
[image]
[image]
I waited for hours but the program still kept me  waiting.
my OS is win10
Someone said I should change to the Agents library, cause download the modules from the default library in China is extremely slow but I don’ t know how .I try to use a vpn but the problem remains. Do anyone know what cause the problem and how to solve it?
[image]

ivankravets, 2021-01-05:
Could you try to install a library from [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli)?

Please type the next in VSCode Terminal

```
pio lib -g install OneWire
```

Does it work? Just for a test.

James-C-Hunt, 2021-01-05:
Yes the PlatformIO CLI works fun. I just create a program and install a library from PlatformIO CLI sucessfully.

[image]

ivankravets, 2021-01-05:
Did you configure a proxy for VSCode? https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support

James-C-Hunt, 2021-01-06:
I noticed this issue before ,and I disabled the Http: Proxy Strict SSL as below
[image]
but I remain the "Http: Proxy" item enpty cause I don't know how to set it. Shall I fill the 'Http: Proxy' item with "http://user:pass@10.10.1.10:3128/"  or I should replace the "user" and "pass" with  my OS user's name and pass word?
[image]
as above?

James-C-Hunt, 2021-01-06:
[image]
I set the 'Http: Proxy' as above .Then I creat a project, and waited for hours but the issue remains.
[image]

ivankravets, 2021-01-06:
Do you have 3rd party antivirus tools in the system?

James-C-Hunt, 2021-01-07:
yes shall I shut it down and have a try? Or there is another way to figgure it out?

James-C-Hunt, 2021-01-07:
My PlatformIO CLI works fine. Do you have any examples or abc instructions of how to use PlatformIO CLI ?  Reading the Docs is a little tough for me : )

James-C-Hunt, 2021-01-09:
The quick start gives a good abc example of how to use it .https://docs.platformio.org/en/latest/core/quickstart.html

closed 2021-01-09 by James-C-Hunt