platformio/platformio-vscode-ide #2298: Error: Error starting process (ENOENT) for all PlatfromIO Tasks

yunuscukran, 2021-01-05:
**Description**

When I try to run any PlatfromIO task from the GUI, I get the following exception. Note that PlatformIO loads and recognizes the project. Moreover running these tasks via the PlatformIO CLI also works. It is only calling them through the GUI that fails.

**Configuration**

I build` platformio-vscode-ide `version `2.1.0` from source code and run it within the Theia IDE, which uses the VSCode API to ingest VSCode extensions
```
 core_version: 5.0.4
 python_version: 3.7.7
 installer_version: '0.3.5'
 OS: Microsoft Windows 10 Home 10.0.18363 Build 18363
```
**Note:**  I come across this problem **only in Windows**. It works in Linux or Mac.

**Exception**
```
process ERROR Error: File not found:
    at new WindowsPtyAgent (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\@theia\node-pty\lib\windowsPtyAgent.js:74:36)
    at new WindowsTerminal (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\@theia\node-pty\lib\windowsTerminal.js:49:24)
    at Object.spawn (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\@theia\node-pty\lib\index.js:28:12)
    at new TerminalProcess (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\@theia\process\lib\node\terminal-process.js:82:41)
    at _createInstance (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\inversify\lib\resolution\instantiation.js:29:12)
    at Object.resolveInstance (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\inversify\lib\resolution\instantiation.js:49:18)
    at C:\Users\...\Desktop\Aarish\windows-build-test\aarish-ide-release\node_modules\inversify\lib\resolution\resolver.js:73:42
    at Object.resolve (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\inversify\lib\resolution\resolver.js:97:12)
    at C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\inversify\lib\container\container.js:321:37
    at Container._get (C:\Users\...\Desktop\Aarish\windows-build-test\project-ide-release\node_modules\inversify\lib\container\container.js:312:44)
task ERROR Error occurred while creating task: Error: Error starting process (ENOENT)
task ERROR Error launching task 'Devices': Error starting process (ENOENT)
```
**Extra Information**

I'm sharing the logs of successful PIO initialization in case they prove useful.

```
root INFO [hosted-plugin: 1204] runCommand C:\Users\...\.platformio\penv\Scripts\python.exe [
  '-m',
  'platformio',
  '-c',
  'vscode',
  'init',
  '--ide',
  'vscode',
  '--project-dir',
  'c:\\Users\\...\\Documents\\PlatformIO\\Projects\\210104-233735-freedom-e-sdk_sifive-welcome'
] {}
root INFO [hosted-plugin: 1204] runCommand C:\Users\...\.platformio\penv\Scripts\python.exe [
  '-c',
  'import os;from platformio.project.config import ProjectConfig;c = ProjectConfig();libdeps_dir = c.get_optional_dir("libdeps");watch_dirs = [c.get_optional_dir("globallib"), c.get_optional_dir("lib"), libdeps_dir];watch_dirs.extend(os.path.join(libdeps_dir, d) for d in (os.listdir(libdeps_dir) if os.path.isdir(libdeps_dir) else []) if os.path.isdir(os.path.join(libdeps_dir, d)));print(":".join(watch_dirs))'
] {
  spawnOptions: {
    cwd: 'c:\\Users\\...\\Documents\\PlatformIO\\Projects\\210104-233735-freedom-e-sdk_sifive-welcome'
  }
}
```

yunuscukran, 2021-01-17:
Any updates on thıs? Still happens with the newest version of platfromio-vscode-ide

ivankravets, 2021-01-20:
Do you use 3rd party antivirus tools in the system?

yunuscukran, 2021-01-22:
Nope, only Windows Defender. 

My initial hypothesis was that the `pio.exe` path was not being created properly at app initialization and the tasks created at [`toVsCodeTask()` ](https://github.com/platformio/platformio-vscode-ide/blob/de60f597f36299e4f3811c00ba84d50b50a5661a/src/tasks/project.js#L152) were having incorrect PATH data, but logging shows that path creation is correct.

I have also tried bumping up the version. One observation is that when I built from the source and attached it to VSCode, Everything works fine in VSCode...

Note that I change the source code as I mentioned in https://github.com/platformio/platformio-vscode-ide/issues/1802#issuecomment-710683116

My current hypothesis is that somehow the way the tasks are created are working fine in VSCode, but not working with Theia (only in the Windows case) and that this is a Theia/VSCode API compatibility issue. I'll let you know as I learn more.

ivankravets, 2021-03-20:
Have you resolved this issue?

closed 2021-05-27 by stale[bot]