platformio/platformio-vscode-ide #2303: Strange "extra_scripts" behavior

gsieben, 2021-01-10:
I have a workspace with two projects (A&B). In one of the projects (A) I have an extra script that automatically increments the build number on build. This works quite well. But now I have the problem when I compile the second project (B) without extra script. Despite that only this project (B) is compiled, the exstra script in project (A) is still executed and the build number is incremented. This is of course not wanted. What am I doing wrong?

[platformio]
default_envs = d1_mini

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino

extra_scripts =
    pre:risebuilnumber.py ; Pyton script increasing the build number

lib_deps = bblanchon/ArduinoJson @ ^6.17.2
           Wire
           knolleary/PubSubClient @ ^2.8
           fastled/FastLED @ ^3.4.0

ivankravets, 2021-01-10:
Thanks for the provided project. This is not a bug. When you open VSCode - it fetches metadata from each project in the workspace to provide code completion and other things.

You can ignore these calls by checking in your extra script

```python

from SCons.Script import COMMAND_LINE_TARGETS

if "idedata" in COMMAND_LINE_TARGETS:
  env.Exit(0)

```

closed 2021-01-10 by ivankravets

gsieben, 2021-01-11:
Thank you Ivan. You have helped me a lot with this. And thank you for this wonderful PlatformIO. I love it!!!!!

ivankravets, 2021-01-11:
Happy coding with PlatformIO! 🤗