platformio/platformio-vscode-ide #2305: Extension issue

joecmay, 2021-01-10:
- Issue Type: `Bug`
- Extension Name: `platformio-ide`
- Extension Version: `2.2.1`
- OS Version: `Windows_NT x64 10.0.18363`
- VSCode version: `1.52.1`

:warning: We have written the needed data into your clipboard. Please paste! :warning:

joecmay, 2021-01-10:
```json 
{
	"messages": [],
	"activationTimes": {
		"codeLoadingTime": 11294,
		"activateCallTime": 3,
		"activateResolvedTime": 0,
		"activationReason": {
			"startup": false,
			"extensionId": {
				"value": "platformio.platformio-ide",
				"_lower": "platformio.platformio-ide"
			},
			"activationEvent": "*"
		}
	},
	"runtimeErrors": []
}
```

ivankravets, 2021-01-20:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2253

closed 2021-01-20 by ivankravets

busunrise, 2021-02-23:
In which file shall I put the code regarding the platformio performance issue?