platformio/platformio-vscode-ide #2308: Initializing project gives "PIO Core Call Error" although error text says "Succeeded"

maxgerhardt, 2021-01-12:
See thread at https://community.platformio.org/t/hanging-on-create-project-project-examples/18485. 

Two examples are given where a user initializes a project (for a Arduino Zero, then an Arduino Uno) and the output say

```
PIO Core Call Error: "The current working directory C:\Users\graynomad\Documents\PlatformIO\Projects\210112-210923-arduino-blink will be used for the project.\r\n\r\n
The next files/directories have been created in C:\Users\graynomad\Documents\PlatformIO\Projects\210112-210923-arduino-blink\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\n

Error: Processing uno (platform: atmelavr; framework: arduino; board: uno)\r\n

--------------------------------------------------------------------------------\r\n========================= [SUCCESS] Took 0.21 seconds =========================\r\n\r\nEnvironment Status Duration\r\n------------- -------- ------------\r\nuno SUCCESS 00:00:00.210\r\n========================= 1 succeeded in 00:00:00.210 ========================="
```

although all files are created. 

How does it see that as unsuccessfull? And awhat does it attempt do a compilation? (Jugding from `Processing uno (platform: atmelavr; framework: arduino; board: uno)\r\n`

Graynomad, 2021-01-12:
Correction, there are NO source code files created. Only a few README files
Rob

maxgerhardt, 2021-01-12:
This specific error was reportedly fixed after removing `C:\Users\<user>\.platformio\packages` and platforms folder.

closed 2021-01-12 by maxgerhardt

Graynomad, 2021-01-12:
Yes it seems to be fixed. I'm locked out of the forum Max (new user, too many posts) so can't respond there for another 15 hours.

Rob