platformio/platformio-vscode-ide #2309: Visual Studio Code IDE freezing during build/upload

timkoers, 2021-01-12:
I have encountered a lot of freezes lately, when trying to build/upload the project to an ESP32/ESP8266.
I've created a dump file of the what I think is the main process.

It is really starting to annoy me and I simply can't find a way to narrow the problem down to something. I'm using the latest of VS Code 1.52.1 and Platform IO (Core 5.0.4 and Home 3.3.1)

https://filebin.net/ijstekw72faff599

ivankravets, 2021-01-20:
Have you resolved this issue?

timkoers, 2021-01-20:
Nope I haven't. I have no knowledge about components of Visual Studio Code and how to debug them

ivankravets, 2021-01-20:
What is your OS? Do you use 3rd party antivirus tools?

timkoers, 2021-01-22:
Windows 10 Pro 19042.746 (x64), using ESET Internet Security. I haven't tried to disable the antivirus.

ivankravets, 2021-01-22:
Yes, please check your machine. We don't freeze people.

closed 2021-01-22 by ivankravets