platformio/platformio-vscode-ide #2319: platformio-ide.reopenSerialMonitorDelay has no effect for task "Upload and Monitor" resp. requesting a delay option b/w upload/monitor

chrislro2019, 2021-01-19:
Situation: 
I often use task "Upload and Monitor", but my device (ItsyBitsyM0 with Arduino) needs some time to provide its COM port on my Windows 10 PC after the elf is uploaded. The serial monitor starts immediately after a successful upload, but often it complains with "could not open port 'COM20'", sometimes it just works. When failing, I can start monitoring manually shortly after that (task "Monitor"). So I assume a delay between upload end and monitor start could make it work. I found the option "platformio-ide.reopenSerialMonitorDelay", but it looks like it has no effect on task "Upload and Monitor". This is either intended (then I would request a new option to delay between upload/monitor in milliseconds) or a bug.

Versions:
Core 5.0.4
VSCode 1.52.1
Win10

References:
https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-reopenserialmonitordelay
https://community.platformio.org/t/any-way-to-delay-opening-serial-port-monitor/8229
https://community.platformio.org/t/serial-monitor-delay-not-working/12419

ivankravets, 2021-02-09:
Please re-test with https://community.platformio.org/t/platformio-ide-for-vscode-2-3-beta-help-us-with-testing/19219

chrislro2019, 2021-02-10:
I re-tested with 2.3.0-beta.1, but the option **still does not have an effect on "upload and monitor"** ... from end of build it directly starts the terminal - independent from what is set in "platformio-ide.reopenSerialMonitorDelay". I started with smaller numbers (2000), I even tried a very large number (1000000) to exclude unit issues (micros vs. millis).

ivankravets, 2022-12-22:
See solution at https://community.platformio.org/t/any-way-to-delay-opening-serial-port-monitor/8229/5?u=ivankravets

commit 7a83b0ee (2021-02-02): Fixed issues when the "Upload and Monitor" command didn't terminate the running task and didn't reopen a monitor with delay // Resolve #2266 , Resolve #2319
commit 7e9da600 (2021-02-28): Fixed issues when the "Upload and Monitor" command didn't reopen a monitor with delay // Resolve #2319
closed 2022-12-22 by ivankravets