platformio/platformio-vscode-ide #2320: PlatformIO VSCode IDE ignores selected active folder

tntodorov, 2021-01-20:
In one window I have multiple folders open, which contain PlatformIO projects. When I press the 'Build' icon on the PlatformIO toolbar, it always builds the top project in the workspace list, even when a different folder is selected as active in the toolbar. Please see attached screenshot.

[image]

On the other hand, when I issue `pio run` from a PIO terminal inside the desired project folder, then the correct project is built.

OS: Windows 10 x64
VSCode: 1.52.1
PlatformIO: Core 5.0.4

tntodorov, 2021-01-20:
It might be helpful to note:

As per comment https://github.com/platformio/platformio-vscode-ide/issues/2252#issuecomment-739948535 the IDE seems to re-sync, if a new platformio.ini file is opened and it contains a different environment. However, if both projects have the same environment as in my case (e.g. env:tinypico), then the behavior is still as described in this issue.

ivankravets, 2021-01-29:
Sorry, I can't reproduce this issue. Could you reproduce it multiple times after VSCode restart?

kleinmantara, 2021-02-04:
Same here problem here

Visual Studio Code 
```
Version: 1.52.1
Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
Date: 2020-12-16T16:30:02.420Z (1 mo ago)
Electron: 9.3.5
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Darwin x64 19.6.0
```

PlatformIO
```
Core 5.1.0
Home 3.3.3
```

ivankravets, 2021-02-04:
How did you add projects to a workspace? Using PlatformIO Home or VSCode -> File > Add Folder?

tntodorov, 2021-02-04:
I have actually used both methods.

I created the projects one after another with PIO > New Project (from the home page)

Another time I just open the folder from the Explorer context menu "Open with Code" and then add the other project with File > Add Folder.

It doesn't matter if I create the projects and try using them immediately or restart VSCode in between. It doesn't matter if I just reload the window, or fully close the application and open again. Behavior is same on both Windows and Linux.

The attached zip file contains 3 fresh projects, which are opened in the same workspace. Even when I select the active project to be different from tinypico1, when I hit the build button it's always tinypico1 which gets built. All 3 projects were created with the PIO New Project wizard.

[Projects.zip](https://github.com/platformio/platformio-vscode-ide/files/5929058/Projects.zip)

ivankravets, 2021-02-04:
You need to open any file from a project and make it active in editor.

tntodorov, 2021-02-04:
Then why do I have the "active project" selection in the PIO toolbar? This should take precedence on what editor is active. I might even have an active editor with a file which does not belong to any of the projects in the workspace, but I need to copy some code from.... What would the build button do in that case, build the world?

tntodorov, 2021-02-04:
Yet, even if I follow your instructions:

[image]

closed 2021-02-09 by ivankravets

ivankravets, 2021-02-09:
Please re-test with https://community.platformio.org/t/platformio-ide-for-vscode-2-3-beta-help-us-with-testing/19219

tntodorov, 2021-02-11:
@ivankravets 

Just tested the beta version. Thanks for all the hard work you do and honestly, this is an amazing piece of software!

Now I see 2 selectors (drop-down menus) in the PIO toolbar: "Switch Platform IO Project Environment" and the old one "Click to select the active folder". 
1. The behavior of the old selector has not changed in the beta version. The build button always builds the first folder added to the workspace, irrespective of the active folder change or the active file in the editor.
2. The new selector does what I expect when I change the project environment to any of the available environments in the workspace - this environment is built by the Build button, irrespective of the active file in the editor.
3. Having 2 selectors is just confusing. What is the purpose of the old selector "active folder" if it does not do anything with respect to PIO? What makes a folder "active" or "inactive"?
4. It would be nice, if the active project environment implemented a similar logic to the active folder selector. I imagine the workflow to be something like:
      - the active file in the editor determines the currently active project environment, usually the "Default"
      - if the user selects a different active environment for a given project through the selector, the "Default" gets updated in the platformio.ini file and the new selection becomes the default.
      - if the workspace has multiple folders, then activating a file from a different project in the editor also changes the active project environment in the selector, i.e. the selector follows the default environment of the active file in the editor.
      - the user may manually changes the active project environment selection to a different project, which:
          - if different from "Default" for this project, updates this project's platformio.ini file
          - if the user activates any of the **build**|**upload**|**clean**|**monitor**|**terminal** actions from the PIO toolbar, they will be executed for the selected project environment/folder.
          - if the user activates a different file in the editor, or starts typing - and the typed keys are not intercepted as a keyboard shortcut, or the shortcut is a text altering shortcut like CTRL+A, CTRL+C, CTRL+V, etc. - then the selector reverts to following the active editor from which the active project environment is derived.
      - the "active folder" drop-down selector is removed since it does not add any usability improvement (this is my perception, please correct me if I am wrong)

ivankravets, 2021-02-28:
Are you sure you installed the beta version?

siul72, 2023-09-13:
I got the same issue, but what as causing it is that I had the same env name for the different projects in the .pio file.
By having different name it start to work without issues.

ivankravets, 2023-09-14:
@siul72, could you file a new issue at https://github.com/platformio/platformio-vscode-ide/issues?

siul72, 2023-09-17:
> @siul72, could you file a new issue at https://github.com/platformio/platformio-vscode-ide/issues?

Not sure if it should be considered an issue. For me is just a limitation that by have unique env names is fixed.

commit 13671066 (2021-02-09): Automatically switch to the latest project and environment // Resolve #2365 , Resolve #2344 , Resolve #2320