platformio/platformio-vscode-ide #2324: Debugger no working since v2.1.1

Gurkengewuerz, 2021-01-22:
As already tried to solve [in the forum](https://community.platformio.org/t/programm-samd21-directly/18139) with @maxgerhardt, here is the matching issue.  

[As Max has already described](https://community.platformio.org/t/programm-samd21-directly/18139/17?u=gurkengewuerz):

> A custom Atmel SAMD21E17 chip is supposed to be debugged in VSCode. The platformio.ini and board JSON definition are below.
> 
> When using the latest core and PIO VSCode extension version, the error “Could not find specified task” appears for the “PIO > Debug” target and a GDB “EPIPE” error for “PIO Debug (skip pre-debug)”. See https://imgur.com/a/0PCzeUx.
> 
> The behavior on my computer is that OpenOCD is started (but an error is thrown due to me not having an adapter plugged in), without GDB errors or launch errors.
> 
> No antivirus is used, but instead of the powershell in VSCode, a custom MinGW shell / bash is used.
> 
> The error goes away when downgrading PIO extension to 2.1.0, that’s the interesting part.

I tested all PIO extension up to `2.1.0` and none of them worked, only `2.1.0`. I also tested another PC with the newest core and newest tools/platforms with exact the same behavior (“Could not find specified task” appears for the “PIO > Debug” target and a GDB “EPIPE” error for “PIO Debug (skip pre-debug)”). No matter if Linux or Windows (dual boot). My MinGW shell was also switched to powershell.    
Everything is tested using the SAM Framework.

My `platformio.ini`
```ini
[platformio]
default_envs = control

[common]
monitor_speed = 9600
extra_scripts = pio/name-firmware.py
build_flags = -D MON_SPEED=${common.monitor_speed}

[env:control]
platform = atmelsam
board_build.variants_dir = customBoards
board = control_board
framework = arduino
monitor_speed = ${common.monitor_speed}
build_flags = ${common.build_flags}
extra_scripts = ${common.extra_scripts}
src_filter = +<*> -<counter.cpp> -<function_gen.cpp>
upload_protocol = atmel-ice
debug_tool = atmel-ice
```

My `control_board.json`
```json
{
    "build": {
      "arduino": {
          "ldscript": "flash_without_bootloader.ld"
      },
      "core": "arduino",
      "cpu": "cortex-m0plus",
      "extra_flags": [
        "-DARDUINO_QWIIC_MICRO",
        "-DCRYSTALLESS",
        "-D__SAMD21E17A__",
        "-DARM_MATH_CM0PLUS"
      ],
      "f_cpu": "48000000L",
      "hwids": [
        [
          "0x1B4F",
          "0x0019"
        ],
        [
          "0x1B4F",
          "0xF019"
        ]
      ],
      "mcu": "samd21e17a",
      "usb_product": "Custom ATSAMD21E17A Board",
      "variant": "control_board"
    },
    "debug": {
      "jlink_device": "ATSAMD21E17",
      "openocd_chipname": "at91samd21e17",
      "openocd_target": "at91samdXX",
      "svd_path": "ATSAMD21E174.svd"
    },
    "frameworks": [
      "arduino"
    ],
    "name": "Custom ATSAMD21E17A Board",
    "upload": {
      "disable_flushing": true,
      "maximum_ram_size": 16384,
      "maximum_size": 131072,
      "native_usb": true,
      "offset_address": "0x0000",
      "protocol": "atmel-ice",
      "protocols": [
        "sam-ba",
        "blackmagic",
        "jlink",
        "atmel-ice"
      ],
      "require_upload_port": true,
      "use_1200bps_touch": true,
      "wait_for_upload_port": true
    },
    "url": "https://mc8051.de/",
    "vendor": "mc8051"
  }
```

If you need more information or the whole custom board, please let me know. I'm pleased to help.

ivankravets, 2021-01-23:
Could you comment these lines?
```
[platformio]
default_envs = control
```

Does it work now?

Gurkengewuerz, 2021-01-23:
@ivankravets i'm not at home today. Will let you know later!

Gurkengewuerz, 2021-01-24:
@ivankravets i've commented the two lines out, deleted my `.vscode` folder, restarted VSCode, `Rebuild IntelliSense Index` and started **PIO Debug**. **PIO Debug** gave me no error in first place. After running it and building the debug version it stopped with _GDB: Error: write “EPIPE”_ again. **PIO Debug (skip pre-debug)** give me instant _GDB: Error: write “EPIPE”_.
I've recorded it [here](https://user-images.githubusercontent.com/10966337/105618670-4cf83f00-5dea-11eb-9e72-5c4355a229be.mp4).

julianobst, 2021-02-09:
Hi,
I was struggling with exactly the same problem.
While watching the video of @Gurkengewuerz, I mentioned that we both have installed the "OpenAPI (Swagger) Editor" plugin.
After disabling the OpenAPI plugin and reloading VS Code, debugging works like a charm.
@Gurkengewuerz could you give it a try and may confirm that the problem is related to the OpenAPI plugin?

Gurkengewuerz, 2021-02-09:
oh wow. i disabled the "OpenAPI (Swagger) Editor" extension and reloaded everything and now everything works like a charm! So i can confirm the behaviour from @julianobst.

maxgerhardt, 2021-02-09:
o_o. Magic how that extension can interfere with stuff. I'll definitely add this to my internal notes and maybe @ivankravets can add this to the FAQ.

ivankravets, 2021-02-28:
It looks like a bug of that extension. It overwrites system environment variables. I added it to the conflicted list.

closed 2021-02-28 by ivankravets

alsaibie, 2021-04-24:
I can confirm that another VSCode extension was causing this issue with the Debugger. So, definitely worth disabling other extensions in the Workspace to troubleshoot this issue.

commit 09ad05ff (2021-02-28): Added "OpenAPI (Swagger) Editor" extension to the conflicted list // Resolve #2324
commit b2d1d1bc (2021-03-01): Added "OpenAPI (Swagger) Editor" extension to the conflicted list // Resolve #2324