platformio/platformio-vscode-ide #2325: PlatformIO can't find files on Windows because of case sensitive paths

ilendemli, 2021-01-23:
Running the Build action fails on Windows because the path on with the PlatformIO Core is installed has uppercase characters in its paths.

```
pio.exe --version
PlatformIO Core, version 5.0.4
```

```
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/tzapu/WiFiManager.git#development
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden:
  File "D:\vscode\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
```

Files and folders are present but have an uppercased letter:
```
D:\VSCode\.platformio\penv>dir
 Datenträger in Laufwerk D: ist Data
 Volumeseriennummer: 18F6-8265

 Verzeichnis von D:\VSCode\.platformio\penv

23.01.2021  13:27    <DIR>          .
23.01.2021  13:27    <DIR>          ..
23.01.2021  13:27    <DIR>          Include
23.01.2021  13:27    <DIR>          Lib
23.01.2021  13:27                17 pip.conf
23.01.2021  13:27                93 pyvenv.cfg
23.01.2021  13:27    <DIR>          Scripts
23.01.2021  13:33               218 state.json
               3 Datei(en),            328 Bytes
               5 Verzeichnis(se), 2.211.667.341.312 Bytes frei
```

ivankravets, 2021-01-23:
Could you provide more details? Over 1,000,000 developers use PlatformIO on Windows and don't have problems with "PlatformIO can't find files on Windows because of case sensitive paths"

ilendemli, 2021-01-23:
So I am going to describe what steps I took:
1. Downloaded Visual Code Studio (the zip, but happens with the installer aswell), added the data folder for portable mode
2. Opened VSCode, Extensions and Searched for PlatformIO IDE pressed install.
3. Core gets installed, but unpacked files have uppercased letters (don't know if that is really the issue)
4. Downloaded and unzipped https://github.com/empathyco/labs-slack-bot
5. Added Project to PIO
5. Cleaned 
7.  
<details>
<summary>Errors out with File not Found:</summary>

```
> Executing task: D:\VSCode\.platformio\penv\Scripts\pio.exe run --target clean <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Tool Manager: Installing git+https://github.com/espressif/arduino-esp32.git
Error: Traceback (most recent call last):
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "d:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "d:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "d:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\vscode\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "d:\vscode\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 143, in cli        
    is_test_running,
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 79, in process   
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 60, in run
    self.autoinstall_runtime_packages()
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 69, in autoinstall_runtime_packages
    self.pm.install(self.get_package_spec(name))
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55, in new
    src_dir, remote_url, tag, silent
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "d:\vscode\.platformio\penv\lib\site-packages\platformio\proc.py", line 113, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "D:\VSCode\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "D:\VSCode\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "D:\VSCode\.platformio\penv\Scripts\pio.exe 'run', '--target', 'clean'" terminated with exit code: 1.
```
</details>

8. Checking for `D:\VSCode\.platformio\python3\lib\subprocess.py` -> `D:\VSCode\.platformio\python3\Lib` Lib is uppercased in explorer: 
[image]

ilendemli, 2021-01-23:
1. Opened PIO Home
2. New Project
3. Project Name: Project, Board: DOIT ESP32 DEVKIT V1, Framework: Arduino
4. Selected Libraries from PIO, searced for WebSockets and added Websockets by Markus Sattler to project
5. Tried building it:
<details>

```
> Executing task in folder 2FA: D:\VSCode\.platformio\penv\Scripts\pio.exe run <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebSockets> 2.3.3
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\lib955\WebSockets\SocketIOclient.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSockets.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSocketsClient.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSocketsServer.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-matrix.c.o
In file included from .pio\libdeps\esp32doit-devkit-v1\WebSockets\src\SocketIOclient.cpp:8:0:
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-misc.c.o
.pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSockets.h:189:18: fatal error: WiFi.h: No such file or directory
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\esp32-hal-psram.c.o

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:WiFi.h
*
**************************************************************

compilation terminated.
In file included from .pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSocketsClient.cpp:25:0:
.pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSockets.h:189:18: fatal error: WiFi.h: No such file or directory

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:WiFi.h
*
**************************************************************

compilation terminated.
In file included from .pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSockets.cpp:25:0:
.pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSockets.h:189:18: fatal error: WiFi.h: No such file or directory

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:WiFi.h
*
**************************************************************

compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib955\WebSockets\SocketIOclient.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSocketsClient.cpp.o] Error 1
In file included from .pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSocketsServer.cpp:25:0:
.pio\libdeps\esp32doit-devkit-v1\WebSockets\src\WebSockets.h:189:18: fatal error: WiFi.h: No such file or directory

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:WiFi.h
*
**************************************************************

compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSocketsServer.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib955\WebSockets\WebSockets.cpp.o] Error 1
=============================================================================================== [FAILED] Took 3.04 seconds ===============================================================================================
The terminal process "D:\VSCode\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
```
</summary>

maxgerhardt, 2021-02-01:
The error 

```
Library Manager: Installing git+https://github.com/tzapu/WiFiManager.git#development
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden:
  File "D:\vscode\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
```

is not about that `D:\vscode` file not being able to be found, but the library you're trying to install. The PlatformIO core does not have a fault for case-sensitive paths on Windows. Window's file system in inherently case-insensitive.

As you can see in https://github.com/tzapu/WiFiManager.git ..

[image]

There's no branch or tag named `development`. Use the master branch or a tagged release.

ivankravets, 2021-02-01:
Thanks, @maxgerhardt !

closed 2021-02-01 by ivankravets