platformio/platformio-vscode-ide #2330: Start PIO Home Server

Rudolf1307, 2021-01-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Darwin, 17.7.0, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 19, in <module>
    from platformio.commands.home.helpers import is_port_used
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/helpers.py", line 26
    async def request(  %23 pylint: disable=signature-differs,invalid-overridden-method
            ^
SyntaxError: invalid syntax

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /Users/rudolfnordmann/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/Users/rudolfnordmann/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

menglert2, 2021-01-28:
I am seeing the same Exception on Mint18.  I noticed an update to something on startup of VScode before starting to get this error, but i did not catch what it had updated.

ivankravets, 2021-01-30:
See https://github.com/platformio/platformio-vscode-ide/issues/2335#issuecomment-770232126

You need to have Python 3 >= 3.6 in your system.

closed 2021-01-30 by ivankravets