platformio/platformio-vscode-ide #2334: Global variables are not shown for AVR + avr-stub

maxgerhardt, 2021-01-29:
For the given `platformio.ini`

```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
debug_tool = avr-stub
debug_port = \\.\COM14
debug_build_flags = -O0 -ggdb3 -g3
lib_deps =
    jdolinay/avr-debugger @ ~1.1
```

and creating a simple sketch

```cpp
#include "Arduino.h"
#include "avr8-stub.h"

volatile int lastAnalogRead = 0;

void setup()
{
  // initialize GDB stub
  debug_init();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  lastAnalogRead = analogRead(A0);
  if(lastAnalogRead > 128) 
      digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

the global variable `lastAnalogRead` is not shown

[image]

Although it shows up when doing a `info variables` command in the GDB console, as also see in the output above.

The error does not change when the `debug_build_flags` are removed.

The debugger still shows local variables as normal.

[image]

Using latest PlatformIO home and core version.

vtmarvin, 2021-01-30:
Had the same problem (cpp development for Arduino UNO), watch tooltip stated that it is _"unable to create variable object"_ and I was not able to see any watched variables in UI.
I tried to add some debug flags and to change port from COM3 to \\.\COM3 but with no success.
After playing with platformio.ini a while I gave it up and returned to original configuration I started with and ... it started working at least partially.
[image]

Now I see variable in watches but not in global/local variables.
Variable was and is still present while using `list variables` in debug console

My platformio.ini
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
debug_tool = avr-stub
debug_port = COM3

; GDB stub implementation
lib_deps =
    jdolinay/avr-debugger @ ~1.1
```
It still works after restarting VSC/PIO.

nelsonov, 2021-11-16:
I am having exactly the same problem with both an Uno and a Mega2560.  Is there a workaround for this?

Fr33man85, 2021-12-12:
1+

jacksontong, 2021-12-27:
same problem

jfdzar, 2022-02-09:
Same problem here!

smwoodward, 2022-09-28:
I'm also having the same issue.  I can see it under watch and in terminal when I print the variable.