platformio/platformio-vscode-ide #2336: PIO SVD doesn't handle "derivedFrom=" correctly, when applied to a "register" line, or when the name uses a "%s" placeholder

be-philippe, 2021-01-30:
Specifically, in `~/.platformio/platforms/atmelsam/misc/svd/ATSAMD21G18A.svd`, lines 7716
`<register derivedFrom="PINCFG0_%s">` and 7838 `<register derivedFrom="PMUX0_%s">` are not handled correctly.

For example, the debugger shows correctly the byte value and the bit fields for `PINCFG0_0` and `PMUX0_0`, but `PINCFG1_0` and `PMUX1_0` are show as 32-bit words (overlapping with the following registers).

As the other uses of `derivedFrom=` are at the `<peripheral>` level and without `%s` placeholder (in this SVD file), I can't known which one of these features causes the problem, or if it's their combination.

be-philippe, 2021-01-31:
Actually, `~/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-vscode-debug/dist/extension.js` only handles `derivedFrom=` for `<peripheral>`, not for `<register>`, `<cluster>` nor `<field>`.

I didn't find the sources (non public repository ?), but I did patch it successfuly on my machine (only for `<register>`).

I'm willing to create a PR.

maxgerhardt, 2021-02-01:
Nice find, in that minified `extension.js` file. But also can't find the source (or, non-minified) version for that NPM module. https://www.npmjs.com/package/platformio-vscode-debug and https://www.jsdelivr.com/package/npm/platformio-vscode-debug show the package and package data, but all in minified JavaScript. The "repository" link links to this repo here, but I can't seem to be able to find the source code here (e.g., the `_loadSVD()`) function.