platformio/platformio-vscode-ide #2337: PIO uploading to Teensy 4.0:  no libusb-0.1.so.4

YB17, 2021-01-31:
Hello, 
I'm using Jetson Nano with PIO on VSCode and successfully compiled the code for Teensy 4.0. During uploading the following error occurred:

Uploading .pio/build/teensy40/firmware.hex
teensy_loader_cli: error while loading shared libraries: libusb-0.1.so.4: cannot open shared object file: No such file or directory
*** [upload] Error 127

I did a bit search and installed libusb-1.0 which is probably not the right one, and also run pio update. Problem remains. Can anybody help? Thanks in advance!

maxgerhardt, 2021-02-01:
Have you tried `sudo apt-get install libusb-0.1-4`? Seems like it needs libusb-0.1 in the system to work properly. You can also `sudo apt search libusb-0.1` to find packages which might work for installation.

YB17, 2021-02-01:
Thanks for the reply, I did try to look for libusb-0.1-4 but after apt update I still can not locate it. Maybe I didn't add the right repository?

YB17, 2021-02-01:
Also there is nothing found when using apt search

maxgerhardt, 2021-02-01:
Well that's weird, blog posts like https://www.infinite8institute.com/zed-nano-setup clearly show that that command is supposed to work specifically on a Jetson Nano. Same on [here](https://gitlab.rlp.net/-/snippets/1150#zed-camera). Another [chinese post](https://blog.csdn.net/weixin_43046653/article/details/93591980) talks about compiling the libraries yourself.

Can you please test the following (self-compiling `teensy_loader_cli`) on the Jetson Nano

```
sudo apt-get install libusb-dev
git clone https://github.com/PaulStoffregen/teensy_loader_cli.git
cd teensy_loader_cli
make
# copy compile output
cp teensy_loader_cli ~/.platformio/packages/tool-teensy/.
```

YB17, 2021-02-01:
Thanks for the fast reply! I checked the first link and indeed it should work. Probably because that I changed the apt source to Chinese local sources since I'm from China and indeed it is much faster for me to update things this way. I'll try to use the default repo and see if it works or not, and if not I'll try the self-compiling method after! 

I'll get back later once I have some results.

YB17, 2021-02-01:
I am writhing this on my nano now. I checked my source.list and the default one is there I believe:
deb http://ports.ubuntu.com/ubuntu-ports/ bionic restricted

and I tried to download it from the Ubuntu package Download site:

https://packages.ubuntu.com/bionic/arm64/libusb-0.1-4/download

but there is nothing listed. I guess maybe the library is obsolete?

maxgerhardt, 2021-02-01:
Interesting. The page https://packages.ubuntu.com/bionic/libusb-0.1-4 shows the differnt architectures, but only `amd64` and `i386` have a download link, **yet** they still list `arm64`.

However, looking at https://ubuntu.pkgs.org/18.04/ubuntu-main-arm64/libusb-0.1-4_0.1.12-31_arm64.deb.html, there is an actual download link for Ubuntu 18.04 Bionic + ARM64 version: http://ports.ubuntu.com/pool/main/libu/libusb/libusb-0.1-4_0.1.12-31_arm64.deb

Can you manually 

```
http://ports.ubuntu.com/pool/main/libu/libusb/libusb-0.1-4_0.1.12-31_arm64.deb
sudo dpkg -i libusb-0.1-4_0.1.12-31_arm64.deb
sudo apt install -f
```
and retry?

Otherwise you should try compiling yourself per above.

YB17, 2021-02-02:
Thanks a lot! the library is installed. But now I'm facing another problem where the VSCode tries to reinstall PIO automatically when I re-opened it, and failed with the following message:

Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
Failed to install PlatformIO IDE.

There is a pop-up small window at the bottom-right corner indicating the process:
PlatformIO: IntelliSense Index Rebuild

Since this is another problem, I've already made a new issue about it. I'll close this one then.

closed 2021-02-02 by YB17