platformio/platformio-vscode-ide #2338: Installation Manager

XINBUTAIYANG, 2021-01-31:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.0.9
  System: Windows_NT, 6.1.7601, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\Administrator\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\node_modules\platformio-node-helpers\dist\index.js:1:34077)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async t.value (c:\Users\Administrator\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\node_modules\platformio-node-helpers\dist\index.js:1:37257)
	at async Promise.all (index 0)
	at async c:\Users\Administrator\.vscode\extensions\comp2300-anu.comp2300-platformio-ide-2.0.9\dist\extension.js:1:5800
  ```

maxgerhardt, 2021-02-01:
>comp2300-anu.comp2300-platformio-ide-2.0.9

Seems like this extension is interfering with the installation. Do you have any other plugins installed in VSCode?

maxgerhardt, 2021-02-02:
Actually you may try and install the PlatformIO core before installing the VSCode extension for it, see https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux. 

Maybe you also need `sudo apt install virtualenv ` or `python3-virtualenv `.

ivankravets, 2021-02-02:
See https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

closed 2021-02-02 by ivankravets