platformio/platformio-vscode-ide #2346: Start PIO Home Server

Rudolf1307, 2021-02-02:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Darwin, 17.7.0, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1254, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1297, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 64, in get_command
    mod = __import__(mod_path, None, None, ["cli"])
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 19, in <module>
    from platformio.commands.home.helpers import is_port_used
  File "/Users/rudolfnordmann/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/helpers.py", line 26
    async def request(  %23 pylint: disable=signature-differs,invalid-overridden-method
            ^
SyntaxError: invalid syntax

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /Users/rudolfnordmann/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:5625
	at ChildProcess.c (/Users/rudolfnordmann/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

elheck, 2021-02-03:
I have the same issue. I think it is due to platformio suddenly using python2 when it should be using python3.
I get this error even though my default python is python3.8.

ivankravets, 2021-02-05:
Could you try to run the installer script manually?
https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux

nurquhar, 2021-02-05:
My Duplicate :  https://github.com/platformio/platformio-vscode-ide/issues/2351

I had this issue also which led me to ask this question on Python:
https://community.platformio.org/t/how-to-fix-pio-command-not-working/19084

Essentially what fixed it for me was uninstalling Python 2.7, installing 3.9 from the MS App store and running the get-platformio.py script.

ivankravets, 2021-02-05:
PlatformIO IDE for VSCode comes with built-in Python. It seems you have 3d party antivirus tools that block PlatformIO.

nurquhar, 2021-02-05:
Not sure where antivirus comes into it. I am up and running again but did not need to touch my Nortorn 360 AV.

ivankravets, 2021-02-05:
Yes, espionage 3rd party antivirus tools make a problem for normal software. It seems Nortorn 360 AV blocks `USERPROFILE/.platformio/python3`. That is the reason why you faced this issue.

woyzeck, 2021-02-05:
Same here, but on a mac without any antivirus.

ivankravets, 2021-02-06:
See https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5

elheck, 2021-02-07:
@ivankravets  This didn't work for me either. See [this comment](https://github.com/platformio/platformio-vscode-ide/issues/2346#issuecomment-772472008)

ivankravets, 2021-02-07:
@elheck could it be linked with https://github.com/platformio/platformio-core-installer/issues/85?

elheck, 2021-02-08:
Thank you @ivankravets that solved it for me.

woyzeck, 2021-02-10:
I can confirm that the steps outlines in https://docs.platformio.org/en/latest/core/migration.html#drop-support-for-python-2-and-3-5 solved the problem on my mac.

closed 2021-03-03 by ivankravets