platformio/platformio-vscode-ide #2363: Add protection for infinate index rebuilding

ivankravets, 2021-02-08:
Allow maximum 5 auto-rebuilding per 1 minute. After this limit - stop the rebuilding process. Provide warning message

closed 2021-02-09 by ivankravets

rrayman, 2021-11-29:
For those of us that are new to this environment how do you change to the lib folder?

DeathmannX, 2021-12-23:
> For those of us that are new to this environment how do you change to the lib folder?

Any updates how to do that?

ivankravets, 2021-12-24:
Please check the extra scripts that you don't have recursion. Do you have in "platformio.ini" `extra_scripts` option?

techniker, 2022-02-14:
Same problem here...

rasmousein, 2022-02-23:
What ?

techniker, 2022-02-23:
Never mind, fixed itself.

the78mole, 2022-02-26:
> Never mind, fixed itself.

Could you write down, how you solved it? I have the same Problem.

Blaseee, 2022-04-28:
I faced the same problem, but now the status of the build is SUCCESS.

Here is my procedure, but I do not guarantee it. And whether the Marlin is necessary from step 1, I do not want to check afterwards.

Step 1: I installed the "Auto Build Marlin" plugin and loaded the current "platformio" as a project over Open a Marlin 2.x Folder.

Step 2: In "Platform IO" I now loaded my build that I wanted to create (this was also downloaded and not created by me). In the "platformio.ini" I had to adjust a lib_deps path, because the required download was no longer available at the address, but an alternative was found and the link in the ini was adjusted.

Step 3: Now I closed the program, rebooted the computer and pressed the build button directly after opening the program. After that everything went through without errors.

As I said, this worked for me and my build, if there are other adjustments to be made with the same basic problem, I don't know.

I hope I could help.

pmead7399, 2022-05-25:
seems utterly ridiculous to a guy like me that is dealing with errors then trying to compile again. Kind of feels like I entered the wrong code into someone's iphone and I'm being punished for it.

robertstuart, 2022-05-27:
Sorry.  I am having the same problem and need some help.  I am a new user of PlatformIO. As for the possible causes:
1. I have made no changes to the standard installation.
2. I turned off my anti-virus with no change in behavior.
3. I have made no changes to the ini file.
Finally, I don't know how to "Rebuild C/C++ Index".  Could you give me some more detailed instructions?  Thanks.

ivankravets, 2022-05-27:
I've just updated the first recommendation regarding libdeps/lib folders.

@robertstuart, could you share your `lib_deps` list from the `platformio.ini`? it looks like PlatformIO reinstalls libraries on every build. This triggers IDE to rebuild C/C++ index because new libs are detected in `.pio/$env/libdeps` directory. 

Could you check `.pio/$env/libdeps`? Do you see recently added folders?

robertstuart, 2022-05-27:
Here is the platformio.ini:
  bob@roberts-air SevenPotatoe % cat platformio.ini

    [env:teensy40]
    platform = teensy
    board = teensy40
    framework = arduino
    lib_deps = 
	    pololu/Tic@^2.1.1
	    sparkfun/SparkFun 9DoF IMU Breakout - ICM 20948 - Arduino Library@^1.2.9

The libdeps folder shows both libraries, neither of which was recently added.

ivankravets, 2022-05-28:
@robertstuart , does this project work well for you [robertstuart.zip](https://github.com/platformio/platformio-vscode-ide/files/8791590/robertstuart.zip) ?

robertstuart, 2022-05-28:
No.  The project you sent me  shows the exact same behavior.  This problem started recently - probably after the last update.

Thanks for helping with this...

ivankravets, 2022-05-28:
See https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system

1. Please remove global PIO Cores ( see docs above )
2. Remove `%USEPROFILE%/.platformio` folder and restart VSCode. Wait when all packages are installed.

Did it help?

robertstuart, 2022-05-28:
I don't understand 1. but I don't think I have CLI or any other installations.  I did 2. and after the re-installation, the problem still exists.

HelgeBoe, 2022-05-29:
Pretty much the same issue here...
Used VSCode + platformio for quite some time without any problems on my MacBook to do some ESP32 programming. Couple of days ago IntelliSense stopped working (without me doing any changes to my projects, neither to the code nor to any ini files etc.), I did a re-installation, basically as suggested above (uninstalling the extension in VSCode, deleting Met .platformio-folder, and then again installing platformio), but the problem still remains the same.

robertstuart, 2022-05-29:
It's probably irrelevant but I note that both us are using a MacBook.

MassimilianoMontagni98, 2022-05-29:
Same issue and I'm using two MacBookPros an Intel i7 and a M1 max, I already did a full clean installation of Platform.io but the issue is still present

ivankravets, 2022-05-30:
Please open PlatformIO Core CLI (terminal icon) on the bottom status bar), and type `pio system info`. Share an output here. Thanks

MassimilianoMontagni98, 2022-05-30:
@ivankravets  here the output

```sh
--------------------------  -----------------------------------------------------------
PlatformIO Core             6.0.1
Python                      3.9.12-final.0
System Type                 darwin_arm64
Platform                    macOS-12.3.1
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/massimilianomontagni/.platformio
PlatformIO Core Executable  /Users/massimilianomontagni/.platformio/penv/bin/platformio
Python Executable           /Users/massimilianomontagni/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       2
Tools & Toolchains          14
--------------------------  -----------------------------------------------------------
```

robertstuart, 2022-05-30:
Here is mine:

bob@roberts-air SevenPotatoe % pio system info
--------------------------  ------------------------------------------
PlatformIO Core             6.0.1
Python                      3.10.2-final.0
System Type                 darwin_arm64
Platform                    macOS-12.4
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/bob/.platformio
PlatformIO Core Executable  /Users/bob/.platformio/penv/bin/platformio
Python Executable           /Users/bob/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          5
--------------------------  ------------------------------------------
bob@roberts-air SevenPotatoe %

HelgeBoe, 2022-05-30:
here we go...

--------------------------  ---------------------------------------------------------
PlatformIO Core             6.0.2b1
Python                      3.8.9-final.0
System Type                 darwin_x86_64
Platform                    macOS-12.4
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/helge/.platformio
PlatformIO Core Executable  /Users/helge/.platformio/penv/bin/platformio
Python Executable           /Users/helge/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       2
Tools & Toolchains          10
--------------------------  ---------------------------------------------------------

foxieproducts, 2022-05-31:
> See https://docs.platformio.org/en/latest/faq.html#multiple-platformio-cores-in-a-system
> 
> 1. Please remove global PIO Cores ( see docs above )
> 2. Remove `%USEPROFILE%/.platformio` folder and restart VSCode. Wait when all packages are installed.
> 
> Did it help?

Nuking ~/.platformio (from orbit) solved this issue for me. macOS.

robertstuart, 2022-05-31:
It did not help. The problem remains.

ivankravets, 2022-05-31:
Very strange issue, only a few people are affected. I could not reproduce it on macOS or Windows (VM). If someone can provide a remote session to debug this issue, I would be thankful. Please email me@ikravets.com

MassimilianoMontagni98, 2022-05-31:
@foxieproducts yeah same outcome of @robertstuart

MassimilianoMontagni98, 2022-05-31:
@ivankravets I just sent you an email from [massimiliano@solutiontech.tech](mailto:massimiliano@solutiontech.tech)

MonsterThews, 2022-05-31:
I have two issues, including this one. I'll see if Multiple Requests continues despite the other issue. I think it does.
My other issue is IntelliSense Rebuilding and Loading Tasks. They start as soon as I open PIO and never quit. I can't upload, build, clean or open Serial monitor while it Rebuilds and Loads Tasks.

ivankravets, 2022-05-31:
It seems we have found a problem. 

1. Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli)
2. Type `pio upgrade --dev`
3. Restart VSCode.

Does it work now?

MonsterThews, 2022-05-31:
There is no change.Rebuilding IntelliSense Index, Loading Tasks

 
 

   
   - Type pio upgrade --dev

ivankravets, 2022-05-31:
Do you see "multiple requests" warning?

MonsterThews, 2022-05-31:
OH!I haven't seen 'Multiple Requests' since the other issue started.VS Code is stuck in Rebuilding IntelliSense Index and Loading Tasks.
It can't generate that error.

ivankravets, 2022-05-31:
@MonsterThews, could you try `pio project init --ide vscode`? Does this command finish?

robertstuart, 2022-05-31:
I ran the "pio upgrade --dev" and restarted.
I no longer get multiple rebuilds and I get no warnings.  This is good news.  Thanks.

MassimilianoMontagni98, 2022-06-01:
I confirm that the issue is solved in 6.0.2rc2

Thanks @ivankravets

MonsterThews, 2022-06-01:
Thank y'all for your help.

trainman75, 2022-06-01:
+1 for 6.0.2rc2 fixing the problem on Mac.
```
PlatformIO Core             6.0.2rc2
Python                      3.10.4-final.0
System Type                 darwin_x86_64
Platform                    macOS-12.5
```
6.0.1 had also been deleting most of the .pio/build directory immediately after building, if it even got that far, so was rebuilding all of the libraries every time.
Thanks for resolving it.

ivankravets, 2022-06-03:
This is VSCode. We don't manage projects in explorer. Please check that you didn't move your project into another folder.

AmberElferink, 2022-08-17:
I am having the same issue. I tried "pio upgrade --dev", but it didn't fix my issue.
I also:
deleted the .pio and .vscode folder, uninstalled platformio, shut down my PC, installed it again and freshly opened the project. I am still getting the Multiple requests to build project issue.

AmberElferink, 2022-08-17:
I solved my issue.
Apparently on uninstall the hidden .platformio folder in C:/Users/<username> (or ~/.platformio for linux) does not get deleted. I uninstalled Platformio closed VScode, deleted the .pio folder in my project, deleted that .platformio folder. Then opened VScode reinstalled PlatformIO and everything worked properly again.

officesteve, 2022-08-17:
I am no longer interested in your information,Please do not reply to this emai for me

thanks

dagnum, 2022-09-27:
Removing %USERPROFILE%/.platformio folder worked for me.

Ductapemaster, 2023-12-30:
Hello, I seem to be having this issue despite being on a later version of PIO than the above commit is for (6.1.11). I found this thread as I received an error in VSC linking me here.

I've attached as much info as I can about my setup. What I have tried so far:
- Confirming I do not have multiple cores installed (I do not)
- Removing the PIO extension from VSC, deleting ~/.platformio, and reinstalling (no change)

Output of `pio system info`:
```
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
--------------------------  -----------------------------------------------
PlatformIO Core             6.1.11
Python                      3.9.6-final.0
System Type                 darwin_arm64
Platform                    macOS-14.0
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/dankouba/.platformio
PlatformIO Core Executable  /Users/dankouba/.platformio/penv/bin/platformio
Python Executable           /Users/dankouba/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          5
--------------------------  -----------------------------------------------
```

my `platformio.ini`:
```
[env:dodec_esp32]
platform = espressif32
board = esp32dev
framework = arduino
monitor_port = /dev/cu.usbserial-*
monitor_speed = 115200
build_flags = 
	-DLV_CONF_SKIP
	-DLV_CONF_INCLUDE_SIMPLE
	-DLV_COLOR_16_SWAP
	-DUSER_SETUP_LOADED=1
	-DST7796_DRIVER=1
	-DUSE_HSPI_PORT=1
	-DPIN_SDA=18
	-DPIN_SCL=19
	-DTFT_MISO=12
	-DTFT_MOSI=13
	-DTFT_SCLK=14
	-DTFT_CS=15
	-DTFT_DC=21
	-DTFT_RST=22
	-DTFT_BL=23
	-DLV_FONT_MONTSERRAT_12
	-DLV_FONT_MONTSERRAT_16
	-DLV_FONT_MONTSERRAT_20
	-DLV_FONT_MONTSERRAT_24
	-DLV_FONT_MONTSERRAT_26
	-DLV_FONT_MONTSERRAT_30
	-DLV_FONT_MONTSERRAT_44
	-DSPI_FREQUENCY=60000000
	-DSPI_READ_FREQUENCY=60000000
	-DSMOOTH_FONT
	-DCORE_DEBUG_LEVEL=5
lib_deps = 
	Bodmer/TFT_eSPI
	lvgl/lvgl@^8.3.9
	fastled/FastLED@^3.6.0
```

Additionally, I have disabled as many extensions as possible to avoid any clashes. All that is running in my workspace:
- PlatformIO
- Microsoft C/C++ (and dependencies)
- Microsoft Serial Monitor
- vscode-pdf

What additional steps do I need to take to debug?

ivankravets, 2023-12-30:
@Ductapemaster, could you open [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type:
```
pio upgrade --dev
```

Restart VSCode.

Now, open again CLI and type `pio system info`. Do you still see this SSL warning?

Ductapemaster, 2023-12-30:
Hey Ivan, unfortunately the warning is persistent:

```
(base) dankouba@KoubapesonalMBP:~/Documents/projects/domecontroller_fw_esp32$ pio upgrade --dev
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Please wait while upgrading PlatformIO Core ...
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
PlatformIO has been successfully upgraded to 6.1.12a6
Release notes: https://docs.platformio.org/en/latest/history.html
Warning! Please restart IDE to affect PIO Home changes
(base) dankouba@KoubapesonalMBP:~/Documents/projects/domecontroller_fw_esp32$ pio system info
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 6.1.12a6!

*******************************************************************************************************************************************
If you like PlatformIO, please:
- star it on GitHub > https://github.com/platformio/platformio-core
- follow us on LinkedIn to stay up-to-date on the latest project news > https://www.linkedin.com/company/platformio/
*******************************************************************************************************************************************

--------------------------  -----------------------------------------------
PlatformIO Core             6.1.12a6
Python                      3.9.6-final.0
System Type                 darwin_arm64
Platform                    macOS-14.0
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/dankouba/.platformio
PlatformIO Core Executable  /Users/dankouba/.platformio/penv/bin/platformio
Python Executable           /Users/dankouba/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          8
--------------------------  -----------------------------------------------
(base) dankouba@KoubapesonalMBP:~/Documents/projects/domecontroller_fw_esp32$ pio system info
/Users/dankouba/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
--------------------------  -----------------------------------------------
PlatformIO Core             6.1.12a6
Python                      3.9.6-final.0
System Type                 darwin_arm64
Platform                    macOS-14.0
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/dankouba/.platformio
PlatformIO Core Executable  /Users/dankouba/.platformio/penv/bin/platformio
Python Executable           /Users/dankouba/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       1
Tools & Toolchains          8
--------------------------  -----------------------------------------------
```

ivankravets, 2024-01-08:
@Ductapemaster , you issue has been fixed. Please remove `/Users/dankouba/.platformio` folder and restart VSCode.

Ductapemaster, 2024-01-08:
Hi team, really appreciate you working on this for me. Unfortunately while the SSL warning has disappeared and I see the Python version has been upgraded, I am still having an issue with the multiple rebuild issue:

[image]

I have tried this on the dev branch as well, but have the same results. My libraries consistently download and then delete themselves, often breaking a compilation in process. I have to get lucky with timing to get my code to build properly.

Let me know what information I can supply that can help debug, thanks!

ivankravets, 2024-01-08:
@Ductapemaster, please 

1. Close ALL VScode windows, projects, folders
2. Download example https://registry.platformio.org/platforms/platformio/espressif32/examples/arduino-blink
3. Unpack, open. Do you still experience this issue?

Ductapemaster, 2024-01-08:
Hi Ivan, I did as instructed and I do not see the same behavior. 

A couple of notes about my issue, in case they are helpful:
- My instance seems to destroy any library folders that I have included (my project has 3). Once those delete, it constantly redownloads them. This is generally what breaks my compilation step. I noted the example you shared doesn't include any libraries, so I am wondering if that may be why it won't trigger the behavior.
- I have a friend who can clone and compile my project fine on his computer, so I suspect it is something with my setup specifically.

ivankravets, 2024-01-08:
Have you read https://docs.platformio.org/en/latest/librarymanager/dependencies.html ? It would be best if YOU do NOT manipulate with libraries in runtime, otherwise, it will trigger all components from the integration to re-init the project.

Ductapemaster, 2024-01-08:
I am not manipulating anything at runtime — I am using 3 libraries from the PIO system, included using the best practices laid out in that support page.

My `platformio.ini` looks like this:

```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:dodec_esp32]
platform = espressif32
board = esp32dev
framework = arduino
monitor_port = /dev/cu.usbserial-*
monitor_speed = 115200
build_flags = 
	-DLV_CONF_SKIP
	-DLV_CONF_INCLUDE_SIMPLE
	-DLV_COLOR_16_SWAP
	-DUSER_SETUP_LOADED=1
	-DST7796_DRIVER=1
	-DUSE_HSPI_PORT=1
	-DPIN_SDA=18
	-DPIN_SCL=19
	-DTFT_MISO=12
	-DTFT_MOSI=13
	-DTFT_SCLK=14
	-DTFT_CS=15
	-DTFT_DC=21
	-DTFT_RST=22
	-DTFT_BL=23
	-DLV_FONT_MONTSERRAT_12
	-DLV_FONT_MONTSERRAT_16
	-DLV_FONT_MONTSERRAT_20
	-DLV_FONT_MONTSERRAT_24
	-DLV_FONT_MONTSERRAT_26
	-DLV_FONT_MONTSERRAT_30
	-DLV_FONT_MONTSERRAT_44
	-DSPI_FREQUENCY=60000000
	-DSPI_READ_FREQUENCY=60000000
	-DSMOOTH_FONT
	-DCORE_DEBUG_LEVEL=5
lib_deps = 
	Bodmer/TFT_eSPI@^2.5.43
	lvgl/lvgl@^8.3.9
	fastled/FastLED@^3.6.0
```

When I compile, it downloads all the libraries:

```
Resolving dodec_esp32 dependencies...
Library Manager: Installing Bodmer/TFT_eSPI @ ^2.5.43
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: TFT_eSPI@2.5.43 has been installed!
Library Manager: Installing lvgl/lvgl @ ^8.3.9
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: lvgl@8.3.9 has been installed!
Library Manager: Installing fastled/FastLED @ ^3.6.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: FastLED@3.6.0 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!
```

and 

```
 *  Executing task in folder domecontroller_fw_esp32: platformio run --environment dodec_esp32 

Processing dodec_esp32 (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing Bodmer/TFT_eSPI @ ^2.5.43
Unpacking  [####################################]  100%
Library Manager: TFT_eSPI@2.5.43 has been installed!
Library Manager: Installing lvgl/lvgl @ ^8.3.9
Unpacking  [####################################]  100%
Library Manager: lvgl@8.3.9 has been installed!
Library Manager: Installing fastled/FastLED @ ^3.6.0
Unpacking  [####################################]  100%
Library Manager: FastLED@3.6.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.5.43
|-- lvgl @ 8.3.9
|-- FastLED @ 3.6.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
...
```

Then at some time during the compilation step, if those downloaded files aren't already in use by the compiler, some process deletes them, often causing reinitialization and downloading of them, showing the above output again. Often the compilation fails because some dependency isn't available when it goes looking for it.

ivankravets, 2024-01-09:
Could you share somewhere your project WITHOUT `src` folder? Have you set custom https://docs.platformio.org/en/latest/envvars.html ?

Ductapemaster, 2024-01-09:
Here is my project without the `src` folder in it. I have not set any custom environmental variables, that I can remember. Is there an easy way for me to check if any are used?

[domecontroller_fw_esp32.zip](https://github.com/platformio/platformio-vscode-ide/files/13878321/domecontroller_fw_esp32.zip)

ivankravets, 2024-01-19:
I can't reproduce your issue. I see a few solutions:

1. You try to disable ALL extension in the VSCode and keeping only PlatformIO + C/C++ extensions
2. You email me@ikravets.com and we set the remote session. We can try to debug this issue together.

ZanderElectronics, 2024-01-27:
I had this issue. Following worked for me:
Delete Folder PlatformIO/Projects/MyProject/.pio/libdeps/MyDeviceName in (Windows) File Explorer / (MacOs) Finder and build. All Depencies are loading again but now it builds without Error.

juanmartin84, 2025-05-21:
Hello, I think I'm having a similar problem.

I'm trying to update some libs.
But for some reason PIO is still installing the old ones plus the new ones.
Then I get errors during compilation.

The red arrows point the old ones.
[image]

below the extract of the platform.ini file

```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip
board = esp32dev
framework = arduino

monitor_speed = 115200
monitor_port = COM[34]; Windows, COM3 or COM4

board_build.flash_mode = dio
board_upload.flash_size = 4MB

board_build.f_cpu = 240000000L ;set clock frequency to 240MHz
board_build.f_flash = 80000000L ; set flash frequency to 80MHz
board_build.partitions = partition_table_HUGE_APP.csv 
board_build.filesystem = littlefs

upload_speed = 921600
upload_port = COM[34]; COM3 or COM4

lib_compat_mode = strict
lib_ldf_mode = chain

lib_deps = 
  bblanchon/ArduinoJson@7.4.1
  ESP32Async/ESPAsyncWebServer@3.7.7
  ESP32Async/AsyncTCP@3.4.1
  ;mathieucarbou/ESPAsyncWebServer@3.6.0
  ;mathieucarbou/AsyncTCP@3.3.2
  knolleary/PubSubClient@2.8
  fbiego/ESP32Time@2.0.6
  adafruit/Adafruit ADS1X15@2.5.0
  northernwidget/DS3231@1.1.2
```

Can somebody give some advice?

Thanks

juanmartin84, 2025-05-22:
I tried removing `/Users/<user>/.platformio` but problem is still there.

```
--------------------------  -----------------------------------------------------
PlatformIO Core             6.1.18
Python                      3.11.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Juan\.platformio
PlatformIO Core Executable  C:\Users\Juan\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\Juan\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          11
--------------------------  ----------------------------------------------------
```
@ivankravets could you give me some advice?

Thanks

juanmartin84, 2025-05-24:
Hello Ivan, after some tests I found the problem. 
There was a library in "lib" folder pushing the installation of those old versions. I removed that lib and problem disappeared. 
Unfortunately, I have prescind of those functions now.

davidlmorris, 2026-01-12:
I've had this issue with Libraries that do not contain a library.json file (where the lib was stored outside the normal src tree using a symlink (lib_ldf_mode = chain)).  The fix was to add a minimal `library.json` file with name and version using the information in the `library.properties` file.

For example:
```
lib_deps = ...
   MIDIUSB=symlink://C:/Home/libraries/lib/MIDIUSB 
   ...
```

Then add C:/Home/libraries/lib/MIDIUSB/library.json containing:
```
{
	"name": "MIDIUSB",
	"version" :"1.0.5"
}
```

I would be neat if PlatformIO could read a `library.properties` (correctly?) if  `library.json` was not present.

commit b9f72be9 (2021-02-09): Added protection for infinite IntelliSense index rebuilding // Resolve #2363