platformio/platformio-vscode-ide #2368: SVD Parser fails for ATMega files with "TypeError: Cannot read property '0' of undefined"

maxgerhardt, 2021-02-10:
See https://github.com/platformio/platform-atmelavr/pull/238 for the SVD files that this occurrs with. Tested with the Arduino Uno json file as described.

[image]

On a sidenote, when loading the SVD files in CLion (in the "Peripherals" tab for a "PlatformIO Debug" target), it doesn't complain

[image]

And can read out some registers / addresses, but fails to read out the I/O registers. Probably becauase AVR memory spaces are weird and it's using the wrong commands to read them from the SRAM space (and not the Flash space) or `avr-gdb` has a bug or `avr-stub` has a bug. 

[image]

EvilMav, 2021-06-07:
It was obvious it is missing some element when parsing the XML of the SVD. By trial-and-error in comparison with a working SVD I've found that the error disappears when something like
```
      <addressBlock>
        <offset>0xNN</offset>
        <size>0xNN</size>
        <usage>registers</usage>
      </addressBlock>
```
Is included under `<peripherial>`, which seems missing from the generated files (or at least the one I've been playing with).

devilman3d, 2022-01-04:
This is also presents a problem debugging on RP2040 hardware: [https://github.com/platformio/platform-raspberrypi/issues/21](url).

If it helps, I have been hacking on that minified extension.js and narrowed the problem to strange indexing in loadSVD and parse* in PeripheralTreeProvider. It looks to be taking the first character of the input strings rather than the strings themselves. Is it expecting an array? Also, the function that parses integers depending on prefix needs validation on the input.

I vote for #2587...