platformio/platformio-vscode-ide #2373: Active build project not updated when opening file from different project

Bascy, 2021-02-12:
Platformio-ide-2.3.0-beta.1
VSCode 1.53.2
Testing #2276 

When opening a file from a different project in a workspace, the active project shown in taskbar is not changed. Clicking `Build` in taskbar starts building wrong project (the one displayed as active project)

No errors in console log.

[image]

maxgerhardt, 2021-02-15:
The filename `platformio.ini` is written in cursive. That indicates that you have only single-clicked on the file, which VSCode interprets as "peeking" into the file I think. If you double-click on the file, it should not be cursive anymore. Is the behavior the same then?

Bascy, 2021-02-15:
When double clicking, or using Ctrl-K Enter combination, the active project is NOT changed.
When clicking on the active project indicator and selecting from the list, the active projet IS changed.

maxgerhardt, 2021-02-15:
Indeed, I get the same behavior. Should be added to the milestone for 2.3.0 imo.

ivankravets, 2021-03-03:
2.3.0 is out! Please update the PlatformIO IDE extension to the latest version.

closed 2021-03-03 by ivankravets