platformio/platformio-vscode-ide #2382: Could not create new Project inside PIO Home in VS code (and workaround)

mhai97, 2021-02-19:
[image]
It is almost impossible to create new Project or import Arduino project inside VS code.
The only workaround I can find is open http://127.0.0.1:8008/ and create project from there, which almost immediately finish.

weypro, 2021-02-20:
I have the same issue, even if I use my vpn.

ivankravets, 2021-03-03:
Could you update the PlatformIO IDE extension to the latest 2.3.0 version and try again? Does it work now?

rajithavk, 2021-03-13:
Doesn't work with the latest (2.3.0) either. Literally, right now portfolio is defunct on windows 10.

ivankravets, 2021-03-14:
@weypro  @rajithavk  what is your OS (distributive)?

ivankravets, 2021-03-22:
Please reopen if you still need help

mhai97, 2021-03-23:
I'm using the latest version on Linux Mint 20.1. It's still a problem and I
still have to create project via $pio home

rajithavk, 2021-03-27:
> @weypro @rajithavk what is your OS (distributive)?

I'm on windows 10. 
I uninstalled PIO from VScode and manually removed whatever remaining directories. 
Then Reinstalled from scratch and it resolved the error.

closed 2021-06-04 by stale[bot]

TanDatEmb, 2023-11-04:
Similarly, I also got an error when I created a new project and waited for a long time but it did not complete. When I built the old project before it also didn't work