platformio/platformio-vscode-ide #2395: UNRESPONSIVE extension host, 'platformio.platformio-ide' took 98%

busunrise, 2021-03-01:
I'm using Visual Studio Code together with platformio. All platformio versions higher than 1.8.3 cause the message "UNRESPONSIVE extension host, 'platformio-ide'. Extensions are not running. 

Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041

Profile attached:
[exthost-75bbfa.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/6059921/exthost-75bbfa.cpuprofile.txt)

This error is very annoying, since there is no reasonable error message for the end user. I would appreciate any help.
Thanks a lot!

ivankravets, 2021-03-03:
1. Please update the PlatformIO IDE extension to the latest 2.3.0 version.
2. Remove `%USERPROFILE%/.platformio/penv` folder
3. Remove `%USERPROFILE%/.platformio/python3` folder
4. Restart VSCode.

Does it work now?

busunrise, 2021-03-04:
Thanks for your feedback. I have updated to V 2.3.0. After the restart I got the notification “the extension platformio took a very long time…”. I do not see neither folder %USERPROFILE%/.platformio/penv nor %USERPROFILE%/.platformio/python3. Under %USERPROFILE%/.platformio there are three folders only: .cache, packages, platforms. Therefore, I can’t delete it. I’m using the builtin python38.

Under  <https://docs.platformio.org/en/latest/integration/ide/vscode.html#quick-start> VSCode — PlatformIO 5.1.1a3 documentation Known issues there is mentioned, that the packages manager is unable to install tool (I assume that platformio is meant). What is it referring to?

ivankravets, 2021-03-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2253

closed 2021-03-22 by ivankravets

busunrise, 2021-03-24:
Sorry, but all hints in #2253, #2305, #2312, #2318 to solve this issue do not show proper actions how to solve the high CPU load issue. The explanation under #2253 on dealing with "high CPU load issues" addresses mainly platformio developers rather than users that just want to use platformio.
For the time being I'm using platformio version 1.8.3 This version is running without any problems.

ivankravets, 2021-03-24:
Please contact me@ikravets.com and discuss a remote session that can help to debug this issue. Over 1.5 million developers use PlatformIO and we have 4 reports related to this issue. I hope you understand that this issue is very specific to your machine and installation.