platformio/platformio-vscode-ide #2398: Re-occuring installation error / forced updates

Triangulix, 2021-03-01:
Hi
I'm trying to use Platformio but it regularly decides that it wants to reinstall itself. This happens about every 2 weeks. The re-installation fails because my development system is not connected to the internet and Platformio stops working.

To get Platformio working again I have to connect to the internet and let it do it's re-installation procedure. Then it works again without internet for maybe 2 weeks before the problem repeats.

Is seems to me that Platformio has some kind of forced update mechanism that stops the current version from starting if it wasn't able to check for updates for a period of time.

Why isn't this documented?

ivankravets, 2021-03-03:
Please update PlatformIO IDE extension to 2.3.0.

Triangulix, 2021-03-31:
Did that. Now the issue is back!
When starting VS-Code it just starts re-installing Platformio which fails due to lack of network connection.
Platformio once again enforces an update by ceasing to work until it gets internet connectivity.

ivankravets, 2021-03-31:
Do you use Linux?

Triangulix, 2021-04-01:
I'm using Windows 8 in a VM for microcontroller development

ivankravets, 2021-04-02:
Could it be linked with https://github.com/platformio/platformio-core-installer/issues/644?

Triangulix, 2021-04-03:
Probably not since on my system there is no relationship to system changes.

I can restart VSCode as often as I like. The update is only forced after a couple of weeks without internet connection. Maybe it is exactly a month. I haven't tracked the interval yet since I only use PIO to maintain the firmware of my 3D printer.

I'm using a VM for microcontroller development to have all these fragile toolchains protected from updates and system changes. Usually the VM is has no network connection and gets updates only a couple of times a year when I have 2 days to spare to repair broken toolchains ansd re-apply patches.

ivankravets, 2021-04-04:
Could you reproduce this issue again?

Triangulix, 2021-04-04:
Today? Of course not. But in a month PIO will uninstall again and then kindly ask for internet connection in case I like to use it again.

ivankravets, 2021-04-05:
Strange issue. If there is no Internet connection, PlatformIO does not know about any updates, and should not make any updates. Please inform me if you reproduce this issue again.

Triangulix, 2021-04-06:
PIO doesn't know if there is an update. It just decides to start a procedure after a certain time without internet connection that first deinstalls parts of PIO and then demands internet connection to re-install them. Issue #2406 could have the same cause.
I'll notice you when it happens again in a few weeks.

rconcept, 2021-04-13:
Hi, I just comment a fix in https://github.com/platformio/platformio-vscode-ide/issues/2406#issuecomment-818393939
One problem of re-occuring installation is certainly this line:
https://github.com/platformio/platformio-vscode-ide/blob/ad10ea380a04628614d5450cf61e389a83d831c4/src/main.js#L60
After replacing it with :
`await this.startInstaller(1); `
No more re-installations.

ivankravets, 2021-04-13:
Wow! This is a critical bug!!! We've just fixed it and a new release is ready. Please update to https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.3.2

closed 2021-04-13 by ivankravets

Triangulix, 2021-05-26:
The issue is not fixed! It occured again, as usual.

commit 2f9b00b7 (2021-04-13): Do not raise an exception when checking PlatformIO Core for updates and the Internet is off-line // Resolve #2398