platformio/platformio-vscode-ide #2404: [FR] Documentation for the fact that "Clone Git Project" requires manual installation of Git

DerAndere1, 2021-03-02:
I could not find information that a manual installation of Git (https://git-scm.com/) is required if the user wants to use "Clone Git Project" 

**My steps:** 
-No installation of Git
- while Git is not installed, click on "PlatformIO" in left tool bar. In panel "PLATFORMIO", click "QUICK ACCESS" -> "Miscellaneous" -> Clone Git Project".

**Current result:** 
Nothing happens

**Expected result:**
Pop-up notification that Git has to be installed and added to the PATH System variable in order to use "Clone Git Project"

ivankravets, 2021-03-03:
Is this is a part of VSCode?

DerAndere1, 2021-03-04:
The menu entry "Clone Git Project" is only available after installation of the extension "PlatformIO IDE" in VSCode. 
VSCode:
Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

PlatformIO IDE: Core 5.1.0·Home 3.3.3

DerAndere1, 2021-03-04:
I found the GUI element "Clone Git Project" in source code here:
https://github.com/platformio/platformio-vscode-ide/blob/66a5834be36ab7ed84ccf85c48ae7ea0efa87837/src/views/quick-access-tree.js#L80 . 
Documentation should go below the following section of the platformio-docs repo:

https://github.com/platformio/platformio-docs/blob/develop/integration/ide/vscode.rst#platformio-core-cli